richonguzman/LoRa_APRS_iGate #1: Support for telemetry

tekk, 2023-05-29:
Hi and thanks for an awesome work,
Would it be possible to add telemetry support for instance with BME860 sensor?
Thanks.

casasale, 2023-05-29:
Hi Peter,Theoretically it should be possible and easier to add support for the BME680 environmental sensor considering that Adafruit has an ESP32 library for it.https://www.adafruit.com/product/3660https://randomnerdtutorials.com/esp32-bme680-sensor-arduino/The only challenge would be that it will be lower in the list of things to do, because Ricardo the developer would need the sensor for testing and time to do it, because I imagine he has great  and very interesting new enhancements ongoing that currently have higher priorities.Of course feel free to fork the project and add it yourself, and then you can create a pull request to allow merging it back into the source project. Basically new contributors are highly welcome :) .Thanks and 73 from Chile to the Slovac Republic,Tihomir CA3TSK

tekk, 2023-05-29:
Sorry, my fault.
Of course I meant BME680, like this one:
[image]
But there are many types of course.
I would might be able to make a PR, but I would need help regarding sending the telemetry.
Is there any code that you could prepare if I would fork the repo?
I have several of these sensors in my lab, so it wouldn't be a problem to integrate & test it.

richonguzman, 2023-05-29:
no problem....  let me know what info and how you want to show it and I could to the "sending telemetry" part.

maybe send me a mail directly

tekk, 2023-05-29:
Awesome!

Do you have any ongoing/unfinished work on the project that you would like to merge first?
Otherwise, I can fork the repo and start working on the BME680 sensor support - i.e. reading the data first.

Afterward, you could help me with sending the telemetry data from the sensor on my fork, and after that, we can merge it back into your repo.

Not related to this issue, but a Web UI would be also an awesome feature to have.

For example, my Czech fellow "brother" @ok5tvr has made one on his [Lora RX iGate](https://github.com/ok5tvr/LoRa_RX_Igate), unfortunately, the HTML and all the static web resources are hardcoded into the code and not contained as files on the SPIFFS, but that doesn't matter at the moment.

Here is how it looks:
[image]

I can create a separate Github issue for the Web interface if you like this idea, and I could help with that as well... but not promising myself, as I'm extremely busy these times.

richonguzman, 2023-05-29:
as for forking, GO AHEAD!!! there are lots of ideas to implement, and its nice to have some help! ;)

as for the web UI, I really want this, could you put me in contact with your friend as I'm wanting to have a Web configuration for iGate and Tracker also

tekk, 2023-05-30:
You can find his e-mail address on his [QRZ page](https://www.qrz.com/db/OK5TVR) after logging in.

You can write him, I don't know him personally, I don't even know if he speaks english... just some friends from [our radio club](https://integrac.sk) noticed the repo and have tried his iGate.

Hopefully next weeks I'll manage to make some time to implement BME680 readings and then we can proceed further.

ok5tvr, 2023-05-30:
Hi, I'm Tomas Vlas OK5TVR - this is my Igate project.
Igate - decodes the aprs icon
- decodes the position of the broadcasting station, even from the
compressed GPS position.
- calculates the distance and course to the station
- transmits telemetric data to aprs (number of packets, maximum distance to
the transmitter station, and esp32 core temperature.

richonguzman, 2023-06-24:
so BME280 and BMP280 support has been added, will look at the web  ui and more in a few days ;)

tekk, 2023-06-25:
Awesome @richonguzman sorry for not being helpful at all here, but I was terribly sick.
Will check out the functionality later.
Thanks again!

richonguzman, 2023-12-24:
> Sorry, my fault.
> Of course I meant BME680, like this one: [...]

Hi TEKK

hope you are feeling better,

I added a new branch *adding-BME680* to test and validate the correct use of this module too. Please test

tekk, 2023-12-26:
Hi @richonguzman,
I've just successfully compiled your branch and got the data out of the BME680! Yeeey!
The output needs some adjustments, but it'll go.
Here's what I'm getting:
[image]
The things I've changed:
- in `bme_utils.cpp` line 29: `status = bme.begin(0x77);` ... my sensor was on address 0x77 according to I2C scanner, first I was fighting with wiring as I thought this is the problem, but it wasn't, LOL
- same file, line 141 changed to `humStr = generateHumString(newHum);` as `type` was undefined there
- And of course uncommented BME680 sensor type in header file
After that I was able to get the correct output.
Sorry for not making a PR but I'm happy that I managed to make it work at least...
Thinking about how to fit the kOhms on display (maybe use scrolling or switch the texts once per few seconds...)
Also the kOhms value has to be inserted into telemetry properly, but I'm no keen to APRS protocol, so can't provide any advice on that.
But all in all I think this is a great success! Thank you for making this possible!
73 OM7TEK

richonguzman, 2023-12-26:
nice testing, will do some code improvement and se how the Gas could help out also

tekk, 2023-12-27:
What would be even more awesome, if I could have a BME860 on my tracker, measuring all the values on the go, strapped on a backpack.

richonguzman, 2023-12-27:
we'll you should check the latest branch  of the tracker ;) its running good (not main yet because testing a few more things)

kg5eiu, 2024-03-26:
I tried to add a BME680 to an existing iGate - double checked all the wires and connections here is my diagram:  [https://flic.kr/p/2pFudfu](https://flic.kr/p/2pFudfu)  and when the iGate boots it connects to wifi but then I get a  error on the screen BME/BMP sensor active but no sensor found...  and the iGate locks up:  [https://flic.kr/p/2pFwAaM](https://flic.kr/p/2pFwAaM)  Not much of a coder and I see that OM7TEK got one running with some modifications.   Currently running 2024-03-18  looking for suggestions.

richonguzman, 2024-03-26:
i will try to make a "searchI2Caddresses" procedure to control the changing addressees in BME modules ;)

richonguzman, 2024-06-18:
I2C address search and autoconfig for I2C Wx modules are available on V1.0.3 Web installer

closed 2024-06-18 by richonguzman