richonguzman/LoRa_APRS_iGate #2: Send voltage/Current charge/discharge from the battery in the comment or telemetry

matteoramella, 2023-06-20:
Is it possible to have the voltage and the charge/discharge current on the display and also send them as "Comment" or "Telemetry"? It would be very convenient for an igate/digipeater away from the QTH and battery/solar powered

iw2ejh, 2023-07-04:
> hi, I have already some test for internal battery monitoring would you like to test it?
>
> contact me through mail for further details

Hello,
may I get some info regarding the battery status in the comment? or even better, in the telemetry?
Thanks
Marinoiw2eh

richonguzman, 2023-07-04:
Hi

The voltage of the battery could be added to the comment with no problem

But should the battery be the "internal" 4,2v connection or another pin if you want to monitor an even higher voltage battery with a voltage divider??

richonguzman, 2023-07-04:
The current is a little problematic as there is no info on the internal chip to have this , but it could be calculated by comparing two pins voltage with a 1 ohms resistor

richonguzman, 2023-07-18:
battery info added on comment and also on screen

closed 2023-07-18 by richonguzman