richonguzman/LoRa_APRS_iGate #3: Display always on

OK1FWG, 2023-06-22:
Hi,
great work, thanks.
I am having no luck with the display going blank neither on the TTGO_LoRa32 or Heltec boards. Both display all information correctly, but even if I set the .json file and upload it, no change.

```
"display": {
		"alwaysOn": false,
		"timeout": 4,
		"turn180" : false
	},

```
Looking at the code, the function display_toggle(bool toggle) seem to be okay, but it still simply does not switch the display off after the timeout. This is not good for the OLED displays, as they like to burn out pixels after some time.
Thanks.
Willi, OK1FWG

richonguzman, 2023-06-25:
fixed screen AlwaysOn mode .... I had a small bug going on that was fixed.

now is turning off after "timeout" time like it should

closed 2023-06-25 by richonguzman

richonguzman, 2023-06-25:
you can upload the new firmware (on the firmware folder) now with OTA and all will work ;)