richonguzman/LoRa_APRS_iGate #4: Aprs beacon

IZ2YHL, 2023-06-29:
Hi, regarding the aprs beacon, when "stationmode 1 or 2" is selected, the beacon is only sent via web to aprs.fi (not via radio TX). When "stationmode 3" is selected, the beacon is only sent via radio TX and not via web to aprs.fi. Is this functioning correct?
I noticed that if the aprs.fi passcode is wrong, the same "aprs-is OK" is given even if there is no connection to the server. Thank you.

richonguzman, 2023-06-29:
as for the APRS-IS "OK" :

this should change as soon as the connection is lost into "-" and as soon as reconnected "changed to "OK"

In which case did you check when APRS passcode was wrong?

as for the beacons:
stationMode 1 and 2 are for iGates connected through WiFi to APRS servers , this is why the beacon is through internet and not RF/LoRa
stationMode3 is digirepeater, without needing Wifi as it just repeats what it hears. So this makes only a RF beacon only unit.

IZ2YHL, 2023-06-30:
Hi, thanks for the reply.
For the APRS-IS (in Station Mode 1 and 2) I tried to connect with a deliberately wrong password, and I still got the APRS-IS "OK", on the display and also on the serial monitor: "connecting to APRS- IS ... Connected! (server euro.aprs2.net / Port: 14580)" 
I checked APRS.FI and it is not connected.

For beacons, OK. Since the radio LoRa coverage here in Italy is still very poor, it is possible, for StationMode 1-2, to have the option of using a beacon also in RF/LoRa and for Station Mode 3 (digirepeater), a Wi-Fi beacon (APRS-IS) ? Thank you very much for your availability and the excellent work you are doing.

richonguzman, 2023-06-30:
RF beacon in stationMode1and2 could be done, as WiFi in stationMode3

but, if you have wifi on stationMode3 why not have it as stationMode2 (iGate?)?

will check the passcode validated bug and come back

IZ2YHL, 2023-07-01:
Hi, yes, but simplifying the discussion, could it be possible to choose to have, for beacon mode, the RF LoRa mode, Wifi aprs-is or both (at the same time) for all the station modes? Thank you.

jranma, 2023-07-21:
I kindly support the addition of an option allowing I-Gates to regularly send a beacon.
This for two reasons:
1. In some countries, stations must report regularly on the RF.
2. This makes it possible to know, thanks to the stations which hear our beacon, up to where our I-Gate transmits and is heard.
Best 73, Julien HB9HRD

iw2ejh, 2023-09-24:
Related to the server connection, for what I remember, even if you do not have a valid call and/or valid password, you will be connected  to the server, but the server will not pass any traffic from or to you igate. This is probably the reason why with a wrong password, the connection occurs, (meaning you get internet access, but no data transfer )

do2frk, 2024-01-06:
Eine andere Frage zum Lora Aprs Digirepeater. Im Display auf dem Lora L32 Board  steht "Stations (30min) = 0, ist es möglich die 30min wahlweise zu ändern und auf beispiel 5min setzen?

Mane76, 2024-01-06:
Ja, kannst Du in der Config ändern, Section "Other", Punkt "rememberStationTime". 
Das "Stations = 0" geben bei Dir an, dass in den letzten 30min keine Station gehört wurde. Wenn Du das jetzt auf 5 Minuten setzt ist das Fenster eben entsprechend schmäler.

do2frk, 2024-01-07:
> Ja, kannst Du in der Config ändern, Section "Other", Punkt "rememberStationTime". Das "Stations = 0" geben bei Dir an, dass in den letzten 30min keine Station gehört wurde. Wenn Du das jetzt auf 5 Minuten setzt ist das Fenster eben entsprechend schmäler.

Danke für den Tipp, hat funktioniert. Mir ist jetzt dennoch schleierhaft weshalb mein igate auf aprs.fi nicht angezeigt wird,obwohl passcode 17330 und euro.aprs2.net richtig sind, port 14580

richonguzman, 2024-01-07:
> Danke für den Tipp, hat funktioniert. Mir ist jetzt dennoch schleierhaft weshalb mein igate auf aprs.fi nicht angezeigt wird,obwohl passcode 17330 und euro.aprs2.net richtig sind, port 14580

hallo, dein APRS Passcode ist nicht 17330, finden sie "APRS Passcode Generator" in Google as deinen startet mit 2****

do2frk, 2024-01-07:
> hallo, dein APRS Passcode ist nicht 17330, finden sie "APRS Passcode Generator" in Google as deinen startet mit 2****

Hallo Ricardo, den Passcode habe ich geändert,dennoch keine reaktion auf aprs.fi. 
"aprs_is": {
        "passcode": "2xxxx",
        "server": "euro.aprs2.net",
        "port": 14580,
        "reportingDistance": 10

iw2ejh, 2024-01-07:
try to generate the code without the suffix

closed 2024-05-11 by richonguzman