richonguzman/LoRa_APRS_iGate #6: Igate to digipeater mode if wifi fails

EA2CW, 2023-07-05:
Could it be possible to make the igate works on digi-only mode if the internet connection is down? 
This way the station would be active in either cases.
Thank you,
Mikel EA2CW

Mane76, 2023-07-28:
Hello Mikel,
this feature is already in development, Ricardo and I defined this as "mode 5" which will come soon. 

This will be a combination of Mode 2 (TX iGate) when aprs-is is available and fallback to Mode 3 (Digi on same freq) when aprs-is is not reachable with a comeback to Mode 2 as soon as aprs-is is available again.

This ensures to have maximum availability with minimum usage of frequency.

BR
Manfred

richonguzman, 2023-08-11:
stationMode:5 is now available ;)

closed 2023-08-11 by richonguzman