richonguzman/LoRa_APRS_iGate #8: stationMode Not Valid in Heltec WiFi LoRa 32 V2

SP3WRO, 2023-07-30:
After compiling and uploading the firmware to the Heltec WiFi LoRa 32 V2 device, the following message is displayed:

> ------- ERROR -------
> stationMode Not Valid change it on : /data/igate_conf.json

I have no idea what is incorrectly set in the igate_conf.json file because I didn't change anything.

`{
	"callsign": "SP3WRO-10",
	"stationMode": 1,
	"iGateComment": "LoRa_APRS_iGate",
	"wifi": {
		"AP": [
			{	"ssid": "********",
				"password": "*******",
				"latitude": 52.7013,
				"longitude": 16.3902
			},
			{	"ssid": "WIFI_2",
				"password": "password_2",
				"latitude": 0.0000000,
				"longitude": 0.0000000
			}
		]
	},
	"digi": {
		"comment": "LoRa_APRS_Digirepeater",
		"latitude": 0.0000000,
		"longitude": 0.0000000
	},
	"aprs_is": {
		"passcode": "*****",
		"server": "euro.aprs2.net",
		"port": 14580,
		"reportingDistance": 30
	},
	"lora": {
		"iGateFreq": 433775000,
		"digirepeaterTxFreq": 433775000,
		"digirepeaterRxFreq": 433900000,
		"spreadingFactor": 12,
		"signalBandwidth": 125000,
		"codingRate4": 5,
		"power": 1
	},
	"display": {
		"alwaysOn": true,
		"timeout": 4,
		"turn180" : false
	},
	"syslog": {
		"active": false,
		"server": "192.168.0.100",
		"port": 514
	},
	"other": {
		"beaconInterval": 15,
		"rememberStationTime": 30,
		"sendBatteryVoltage": false
	},
	"bme": {
		"active": false
	}
}`

Mane76, 2023-07-30:
Hello SP3WRO,

could you please check if WLAN credentials are ok (no type in config, no MAC filtering in Accesspoint, ...) ?
This message also appears if WLAN is not connected.

Could you please also check if "passcode" is set correctly in section "aprs_is" of config ?

BR
Manfred

SP3WRO, 2023-07-30:
I checked everything. This is not the first time I run ESP32 with wifi.

richonguzman, 2023-07-30:
the "stationMode" error means you should try to upload the "igate_conf.json" configuration file once again , as the igate did not find it when booting.

Sometimes VSCODE has a bug and you need to turn it off and restart the program once again to be able to upload the configuration file (json) once again.

let me know how it goes

SP3WRO, 2023-07-30:
Thanks! It works!

richonguzman, 2023-07-30:
excelente!

richonguzman, 2023-07-30:
> Thanks! It works! niedz., 30 lip 2023, 13:52 użytkownik Ricardo Guzman (Richonguzman) < ***@***.***> napisał:
> […](#)

try StationMode:2 if you can ;)

closed 2023-07-30 by richonguzman