richonguzman/LoRa_APRS_iGate #10: igate asl elevation in config.json

iemmeti, 2023-09-23:
Hi,
I would suggest to add igate asl elevation in configuration json file (maybe in bme section?), so pressure correction can be made without source editing.
Thank you for your amazing project!
Marco IZ0HCD

iw2ejh, 2024-01-04:
Hi Ricardo,  may be this is a trace to follow?  :https://www.meteocercal.info/forum/thread-1099.html

lemielek, 2024-01-30:
It is partially quite complicated.
A we can have two different localisations because of two different WiFi, so it cannot be single value in bme section, but better two values in wifi.AP sections. Of course you can have two identical values as somebody can have possibility to use two different home wifi networks.
And then recalculation in bme_utils.cpp.
Sorry, I do not know yet how to write patch and do request pull.

richonguzman, 2024-01-31:
this is a solution I will try to implement soon

richonguzman, 2024-06-18:
Height Correction has been added on WebUI on V.1.0.3 web installer

closed 2024-06-18 by richonguzman