richonguzman/LoRa_APRS_iGate #13: Please add support for W32-ETH01 board

invictus737, 2023-10-31:
Hi Ricardo,

great job and many thanks for this project. 
would be great to have a Ethernet connected board with POE possible running igate software. Could you consider porting your code to w32-eth01 board? 
ESP32 cpu and some GPIO to add peripherals. The best addition is the Ethernet port.  Datasheet is available here: https://files.seeedstudio.com/products/102991455/WT32-ETH01_datasheet_V1.1-%20en.pdf

10x/73!
Chris/YO3TCO

richonguzman, 2023-10-31:
If you want to help me test it , I would gladly add this board

Maybe over telegram messaging?

I don't have any of this boards here , this is why

invictus737, 2023-11-03:
Sure, I'm happy to help

richonguzman, 2023-11-03:
Hi could I contact you?

invictus737, 2023-11-03:
on telegram I'm available at search YO3TCO
I've looked for you but no luck

Regards/73!
Chris/YO3TCO

richonguzman, 2023-11-03:
Right now I can write to anyone , please write me at @Richonguzman

tekk, 2023-11-11:
+1 for adding this board. That would be sooo awesome. No more need for Wifi on the hilltops, as there are only ethernet swithes there. Such a great idea, @invictus737