richonguzman/LoRa_APRS_iGate #17: No Syslog with stationMode = 5

AG7U, 2023-11-26:
Syslog events not received by syslog server when stationMode = 5.  Changing stationMode to 2 brings back the syslog events.

richonguzman, 2023-11-26:
Hi,

please download latest version as I think this is corrected now:

when in stationMode5 without Wifi, nothing happens, but with WiFi available it should send to syslog Server

AG7U, 2023-11-26:
stationMode5 results in what looks like a boot loop.  The LoRa APRS (iGate) screen will appear for 7 seconds, then a black screen for 1 second, followed by the same LoRa APRS (iGate) screen again.  That loop/pattern keeps happening over and over and I never see the WiFi screen or the listening screen.  When I change back to stationMode = 2 then everything operates normally.

I did not erase flash or full clean after downloading latest version.  Once I saw the boot loop, I did try erasing flash and performing a full clean but boot loop keeps happening in stationMode = 5.

On a side note, there are several of us in the Denver, Colorado, USA area using iGates and trackers with your code.  We really appreciate what you are doing.  I personally can wait to see your Winlink implementation.  Keep up the good work.

Jerry, AG7U.

richonguzman, 2023-11-26:
thanks for the kinds words

i will come back to SM=5 in a few hours and let you know!

richonguzman, 2023-11-28:
I just tested all here and its working fine stationMode5 with syslog enabled when wifi is available ;)

please update

AG7U, 2023-11-29:
It's been working fine for me for 24-hours.  Thank you.  Ok to close issue.

I sent you $50 via PayPal.  We (several of us in Denver, CO, USA) area appreciate all that you are doing for LoRa APRS development.  Again, I personally cannot wait to see your Winlink implementation.  Keep up the good work.

closed 2023-12-20 by richonguzman