richonguzman/LoRa_APRS_iGate #18: Error stationMode Not valid

AWCowan, 2023-11-27:
All 
using a LILYgo T3 V1.6.1 using it for a iGate.
No matter what frequencies and mode I use its the same error Error stationMode Not valid.
Thanks

Andrew 
config below...

{
    "callsign": "GM0UDL-10",
    "stationMode": 2,
    "iGateComment": "LoRa_APRS_iGate",
    "wifi": {
        "AP": [
            {   "ssid": "xxx",
                "password": "xxxxx",
                "latitude": 57.58481,
                "longitude": -4.19181
            },
            {   "ssid": "WIFI_2",
                "password": "wifi_2_password",
                "latitude": 0.0000000,
                "longitude": 0.0000000
            }
        ]
    },
    "digi": {
        "comment": "LoRa_APRS_Digirepeater",
        "latitude": 57.58481,
        "longitude": -4.19181
    },
    "aprs_is": {
        "passcode": "16566",
        "server": "euro.aprs2.net",
        "port": 14580,
        "reportingDistance": 30
    },
    "lora": {
        "iGateFreq": 433775000,
        "digirepeaterTxFreq": 433775000,
        "digirepeaterRxFreq": 433900000,
        "spreadingFactor": 12,
        "signalBandwidth": 125000,
        "codingRate4": 5,
        "power": 20
    },
    "display": {
        "alwaysOn": true,
        "timeout": 4,
        "turn180" : false
    },
    "syslog": {
        "active": false,
        "server": "192.168.0.100",
        "port": 514
    },
    "bme": {
        "active": false
    },
    "ota": {
        "username":"",
        "password": ""
    },
    "other": {
        "beaconInterval": 15,
        "rememberStationTime": 30,
        "sendBatteryVoltage": false,
        "externalVoltageMeasurement" : false,
        "externalVoltagePin": 34
    }   
}

richonguzman, 2023-11-28:
Hi, please upload in this order:

1) firmware
2) filesystem 

sometimes after firmware upload VSCODE gets crazy. So you turn it off , wait a few seconds and turn it on. This refresh the VSCODE process to identify upload port and this uploads correctly the filesystem later.

Mane76, 2023-11-28:
The same error is shown if connection to WiFi can not be enabled - see your credentials or reduce distance to accesspoint.

AWCowan, 2023-11-28:
Thanks for the help.
I did not understand correctly about file system and firmware.
So after a couple of goes was able to upload the filesystem then it works shows trying to connect to wifi then connects.
Next add screen to Lilygo T beam -AXP2101-V1.2 and setup for a tracker.
Andrew GM0UDL

richonguzman, 2023-11-29:
great, although tracker T-beam can be used without screen and with the cellphone(aprsdroid in android) in the meantime

AWCowan, 2023-11-29:
Tracker now working with screen use Apple here so no way to connect for messaging etc.

AWCowan, 2023-11-29:
Question my Igate does not have gps on it would this cause it not beacon out on RF ?

richonguzman, 2023-11-29:
how is your igate setup?  stationMode is.... ?

what would you like to beacon over RF?

AWCowan, 2023-11-29:
Its setup up like config above. "stationMode": 2,
I thought like old analogue APRS it would send my location and symbol, as when I looked in my T-beam mobile no stations received. useful for showing base station on RF.
thanks
Andrew

richonguzman, 2023-11-29:
for now , when iGate only send their beacons to aprs.fi over TCPIP (internet) not over RF

soon I will mod to enable this if needed

AWCowan, 2023-11-29:
I think it would be a good option to TX or not.
Good for other stations to listen out.
Andrew

radiohound, 2023-11-30:
> Hi, please upload in this order:
> 
> 1. firmware
> 2. filesystem
> 
> sometimes after firmware upload VSCODE gets crazy. So you turn it off , wait a few seconds and turn it on. This refresh the VSCODE process to identify upload port and this uploads correctly the filesystem later.

I'm sorry, but could you explain further how to download the firmware and the filesystem with vscode? I am compiling something, and it's loading to the board, but I can't figure out how to select which one I am compiling. I am getting the same error as the original poster, using the same settings. My said and password are correct.

AWCowan, 2023-11-30:
On my experience so far.

The wifi connects later as when I had the Error stationMode Not valid nothing else would happen.

As soon as the file system was fixed it was seen to look for wifi and connect.

Andrew

richonguzman, 2023-11-30:
great new , I will improve the code and functionality of the igate soon and will add all of this ideas also

AWCowan, 2024-01-04:
richonguzman
Thanks for your software brilliant, we have started building a network up here in the Highlands & Islands of Scotland.
Not sure which way to communicate with you?
Put a donation in from the group.
Some questions
1. Got a version T-beam v3.0 and don't know how to flash it any tips ideas
2. With messaging can it go via an iGate to send to your tracker as it only seems to work between trackers?
3. The licence laws are changing so it will be ok to run TX on iGate up to 5 watts, so I think a RF beacon every 30mins would show up on other iGates as a signal check?
GM0UDL

richonguzman, 2024-01-04:
I wrote you a mail (from QRZ) ;)

closed 2024-01-19 by richonguzman