richonguzman/LoRa_APRS_iGate #23: qAO substring is misplaced in IS-packet, when string contains a ":"

formtapez, 2024-01-16:
Hi, when the string representation of the coordinates contains a ":", the placement of the "qAO"+Callsign substring is inserted at the wrong place in the IS-outputstream, which results in wrong coordinates placed at the other side of the planet.

Generated packet:
`DG0UHK-9>APLC13,qAS,DB0BWL-1:!/3z,qAO,DB0BWL-1:`Q6(D>=BQLoRa-APRSCube`

Should be:
`DG0UHK-9>APLC13,qAS,DB0BWL-1:!/3z:`Q6(D>=BQLoRa-APRSCube,qAO,DB0BWL-1`

Problem here:
https://github.com/richonguzman/LoRa_APRS_iGate/blob/79e88bb03fc1255952885a9c675dffeec10fae2d/src/aprs_is_utils.cpp#L79

formtapez, 2024-01-16:
I got it from  here:
https://aprs.fi/?c=raw&call=DG0UHK-9&limit=500&view=normal
(login needed)

[image]

formtapez, 2024-01-16:
Same packet via syslog:
`Jan 12 14:43:37 nat.db0bwl.ampr.org ESP32 LoRa [APRS] - DB0BWL-1 - LoRa Rx - GPS - DG0UHK-9 / APLC13 / _ / -104dBm / -17.50dB / 434Hz / 52.45038N / 10.82962E / 9.0km
`
btw, im using firmware version 2023.12.07 in which you changed to ` packet.indexOf(":`"))` and then changed it back later.
maybe that bug is already gone in newer versions?

formtapez, 2024-01-16:
As you see (received by DO2DE-10) one packet before and after our incident, the string representation of the GPS-coordinates contains an `:`, at which position our qAO-String was misplaced into.
I got that error also 1 or 2 times before, some weeks ago from the same tracker. At roughly the same position on his way to work.

Maybe the other qAS-thing is a consequential error because the whole string was interpreted the wrong way (like the coordinates, too), and inserted by APRS.fi itself afterwards. So i yould recommend to focus on the `:` 😁

Mane76, 2024-01-16:
I think this is a different issue. Note the :` is a MIC-E encoded paket. We also have seen the issue that sometimes it is appened as comment.
Please update to the latest version as it should be solved. I currently test MIC-E and no problems at all.

richonguzman, 2024-02-25:
the fix was added as the packet upload process was fixed

closed 2024-02-25 by richonguzman

formtapez, 2024-02-25:
Sorry i forgot to come back here to confirm that updating the firmware has completely fixed that problem.
Thanks!