richonguzman/LoRa_APRS_iGate #24: Not hearing APRS

RFRanger, 2024-01-18:
Thanks for the excellent project.  I've loaded the code onto a TTGO LoRa v2.1 (PCB says T3_v1.6) and everything went very smoothly.  The JSON config file is all correct and I didn't change any frequency settings, etc. but it never seems to receive packets I'm sending from any APRS radio.  I've tried from APRSDroid and from my Yaesu FT1-DR (both set to 433.775MHz), but the screen never changes and no APRS info is ever uploaded to aprs.fi.  Is it only meant to work with the TTGO APRS transmitter, or should it be working for me?  I know the board works because I had it receiving satellite packets with TinyGS for many months before switching it to this use.  Thanks again!

RFRanger, 2024-01-19:
Ah, I think I'm not fully understanding.  It seems like maybe the iGate is only meant to work with the associated LoRa tracker, and not with standard UHF transceivers.  After a little digging, it seems like the LoRa iGate works at 300 baud and not 1200 baud like standard APRS.  Is it possible to set it up to work at 1200 baud?  The reading I've been doing makes it seem like changing the spreading factor and Coding Rate can allow it to work at a faster speed, but probably with a reduction in range.

richonguzman, 2024-01-19:
Hi, im sad to bring you this news but they are different protocols

VHF or UHF APRS using AFSK/AX.25 protocol does not talk directly to LoRa (and viceversa)

this is, you need an VHF/UHF AFSK/AX.25 igate to listen to the Yaesy, and this LoRa iGate to listen to LoRa Trackers and other LoRa Stations.

the way that this can connect is when the AFSK/AX.25 igate receives the aprs packet, uploads to aprs-is servers over internet and they would send to LoRa iGate for transmiting the packet to the LoRa Tracker

RFRanger, 2024-01-20:
No worries!  Thanks so much for the explanation - I should have considered that used a different modulation, but it just didn't occur to me.  I think SDRs have spoiled me :)  I'm glad to be learning about this and appreciate your patient reply.

closed 2024-01-22 by richonguzman