richonguzman/LoRa_APRS_iGate #25: Is support for the LILYGO®TTGO T-Echo and similar nRF52 boards possible?

sdhuk, 2024-01-23:
Hi, thanks for the work that you've put into this firmware. It's great.

I'm interested in developing a solar powered digipeater, and the ESP32 based boards are quite power hungry in comparison to the lora boards with lower powered microprocessors. 

Many thanks once again,

Stuart MM7SHX

richonguzman, 2024-01-23:
Hi! 

Do you have a T-echo?

Would love to start porting this firmware into more boards. But for now I don't have any t-echo available for this.

sdhuk, 2024-01-23:
I do yes. I'd be happy to test any firmware you'd like on it. I'd even lend you it but it might be a bit far to post it to you though. :)

sdhuk, 2024-01-23:
Absolutely! Thanks so much for your work. I'm happy to help with testing whenever you're ready. All the best.

mugu57, 2024-03-12:
Hello Ricardo
I would also love if we could use in future the T-Echo. One question which came up here. The best woudl be if you have one T-Echo at your premsies for testing. Can you order one via Amazon or Lillygo or any other online shop right now.
May be the easiest thing would be that we sponsor you one T-Echo in transfering moeney to you via PayPal  and then you have a chance to get it shipped directly to you. That would be much cheaper than shipping it form UK or Germany.
Would that something which would help you?
Just an idea.
vy 73 de Michael DB2GM

travnewmatic, 2024-03-12:
I second this!

I also have a T-Echo and would like to run your software on it.  Happy to help test!

sdhuk, 2024-03-16:
Hopefully it should be fairly straightforward to support the wisblock too if the t-echo works. Happy to help with any testing!

mugu57, 2024-03-18:
Hello sdhuk 
Ricardo and I looked last week a bit inter Wisblock modules.
The only restriction we found that the WiFi capability is missing which would be a serious problem in using it as an iGate.
Do you have any experience with these Wisblock modules.
vy 73 de Michael DB2GM

sdhuk, 2024-03-18:
I think there's an ethernet module available for the wizblocks. My main interest is in using one of these boards as a solar digipeater though.

richonguzman, 2024-03-19:
T-Echo has been ordered ;) (thanks for the  sponsorship!)

I'will close this and continue on the tracker side as soon as it arrives !

closed 2024-03-19 by richonguzman