richonguzman/LoRa_APRS_iGate #28: PSE Help or FAQ for Syslog

AlexeyR4N, 2024-01-30:
Hello !  Please explain to me how to read the syslog data from IGate ? I enabled it in the config file, and I tried connecting to IP 192.168.0.100:514 via chrome.   I'm definitely doing something wrong.
Can you give me some instructions? I'm not a programmer, so this is difficult for me.
Muchos Gracias 73!

Mane76, 2024-01-30:
You do need a syslog server. You can use e.g. a synology nas, a raspberry pi with enabled syslog server or something similar.
The igate connects to the syslog server specified in the config. In a second step you can then access the syslog server and check the content.

richonguzman, 2024-01-31:
I've used this on a raspberry to enable a syslog server to watch all the data, maybe this will help you

closed 2024-01-31 by richonguzman

tekk, 2024-01-31:
How about to write a little walktrough on how to setup Syslog server on RPi and Synology NAS? I'm especially interested in Synology, cause I have one. I would like to gather logs from iGate on a hilltop, and also one far away from me. At least write a sum-up here, please. Thanks!

Mane76, 2024-01-31:
Hello tekk,

ok, here we go: 

**Synology:** 
Follow https://kb.synology.com/en-us/DSM/help/LogCenter/logcenter_server?version=7
with format = bsd, transfer = udp, port = 514, ssl = disabled
You will then find the protocols either in the overview of the log-center or in log-center > protocols > filter to "from other server".

**Raspberry:**
Install rsyslogd on a raspberry if not already present, and check if it is running with "sudo systemctl status rsyslog"
Check if the following two lines are enabled in /etc/rsyslog.conf
```
module(load="imudp")
input(type="imudp" port="514")
```

Create a file "LoraIgateLog.conf" in "/etc/rsyslog.d" with the following content, adopt your logdir and the first three parts of the IP of the iGate

```
$template LoraIgateLog, "/var/log/LoraIgate.log"
if $fromhost-ip startswith "192.168.1." then -?LoraIgateLog
&amp;amp;stop
```

Restart rsyslogd

Mane76, 2024-01-31:
> I would like to gather logs from iGate on a hilltop, and also one far away from me. 

You need to have the possibility that the iGate can connect to the IP of your Synology or Raspberry (either by VPN, port forwarding, beeing already in the same network, ...). The IP you define in the config of the iGate is the reachable IP from the syslog server.