richonguzman/LoRa_APRS_iGate #30: BME280 Sensor: Atmospheric pressure values not correct

oe5reo, 2024-02-08:
I just added an BME280 to my LoRa iGate:
https://en.aprs.fi/info/OE5REO-11

Temperature and humidity seem to work fine ... but the value for atmospheric pressure is completely wrong. It is showing 947 or 946mbar although the pressure right now is about 1000mbar (looked it up on a wx station nearby).

So I am wondering ...

Do I need to calibrate the sensor first?
Do I need to make some changes (perhaps in "bme_utils.cpp")?
Or perhaps is the BME280 faulty and I should try another one?

Thanks for your help

73, Martin
OE5REO

richonguzman, 2024-02-08:
Maybe you need to adjust the height from sea level.

Check this value where you live and adjust it on bme-utils.h or .cpp

oe5reo, 2024-02-08:
Hi Ricardo,

thanks for your quick reply. You mean I should put 470 (I live 470m ASL) here?

[Screenshot 2024-02-08 184122](https://github.com/richonguzman/LoRa_APRS_iGate/assets/97287167/d42ab879-24a9-43f5-adff-7e305d078c33)

richonguzman, 2024-02-08:
Yes

Try with 470 instead of 0 as this is meters height correction

oe5reo, 2024-02-08:
Thank you ... working fine now!

closed 2024-02-08 by oe5reo