richonguzman/LoRa_APRS_iGate #31: Use GPS for iGate coordinates

mweyland, 2024-02-20:
Hello

I am wondering whether this software is able to use a GPS module to determine its position instead of hard-coding the lat/long in the config json. This seems to be a feature in [another implementation](https://github.com/lora-aprs/LoRa_APRS_iGate) by setting `use_gps` but I am not sure how that codebase relates to this project.

Having this feature would enable people to use the iGate with a hotspot, e.g. when on holidays or when visiting random places, but I am not sure what would happen when the GPS module fails to report the current coordinates...

richonguzman, 2024-02-20:
Hi,

this is a great idea and I will come back to do this very soon as the igate is getting big upgrade in few days, then I can add this idea ;)

mweyland, 2024-02-21:
OK this is great, thank you! (Also thank you for adopting my air plane icon in the APRS tracker project)

drtechga, 2024-02-21:
I second this idea

SQ2CPA, 2024-02-21:
That's unfortunately not good idea because then the IGate or Digi will move on aprs.fi map because even if the station doesn't move, the GPS "floats" and sometimes even a long distance

mweyland, 2024-02-22:
Thank you for pointing this out. Indeed, there are circumstances where this would not work as intended. I guess if somebody is on the move (e.g. an iGate on a car) then the  iGate should indeed be moving on the aprs.fi map. In that case, the feature would make sense(?) but this is not the use case I described originally.

If the iGate is stationary during operation, but its position is ultimately not fixed, then SQ2CPA has a good point. How about this: Could there be a menu entry where the GPS coordinates can be polled from the GPS module and used from then on? That would address these concerns while retaining the possibility to change the reported position without flashing...

Lastly, are valid GPS coordinates required for the iGate functionality? The project documentation is very clear on this (you need to enter coordinates) but maybe this is a limitation of the code rather than the protocol?

mweyland, 2024-02-22:
Yes that would be nice, and I think it makes sense to do this in a sensible way, i.e. run this process at startup or upon user request, but not continuously. Otherwise the iGate icon would move around on the aprs.fi map to some extent, which is undesired.

drtechga, 2024-02-23:
I currently use peters Igate as it does have gps. I'm using it as a mobile igate. But like this igate firmware better

richonguzman, 2024-10-21:
GPS in iGate is added in release V2.1

closed 2024-10-21 by richonguzman