richonguzman/LoRa_APRS_iGate #37: OTA password issue

iw2ejh, 2024-02-27:
With latest version (web interface) 2024-02-25  when I go to page <ip-of-iGate>/update/ for OTA update,  I was asked for a password, but the this was not accepted. 
The password is set in the previous page (web interface). I also tryied with a blank password, but get the same behavior.

richonguzman, 2024-02-27:
can you delete any character on the OTA part of the configuration Web Page?

iw2ejh, 2024-02-27:
Yes, I did that too. no username no passord      
then I go on  the OTA page, and I'm asked for a password,  I let all blank and I'm asked again and again for password

richonguzman, 2024-02-27:
maybe cleaning/deleting the browser cache will help you with it

iw2ejh, 2024-02-27:
I used a new browser just for that page, but seme behavior

richonguzman, 2024-02-27:
SAVE the configuration, it will download a .json file

open it in textEditor app to see if a character is saved there as "user"+"password"

PS: which IP are you using for it?

iw2ejh, 2024-02-27:
from the saved file (backup) : >         "ota":{"username":"","password":""}}
I'm connecting to 192.168.1.230 that is assigned to the iGate in my network

richonguzman, 2024-02-27:
so the " 192.168.1.230/update " page does not open or asks for password?

iw2ejh, 2024-02-27:
the page "192.168.1.230/update does not open (blank page) and is asking for a user name and password in a popup. If I close the page using Abort, the page give " HTTP ERROR 401 "
[image]

iw2ejh, 2024-02-27:
[image]

iw2ejh, 2024-02-27:
even if I put some number like Username  1    password   1,    I got the same behavior

iw2ejh, 2024-02-27:
[image]

[image]

[image]

Same behavior

iw2ejh, 2024-03-01:
I took down the igate and flashed it again (VSC) with the empty conf.json  file and using the webpage to configure everything work now.  
I still have a couple of questions: 
1- Can we still flash the new version using  VisualStudioC with the conf.json filled with data, or should we aways flash it with the conf.json file empty and go through the local hotspot for the first config and then continue config via web interface??
2- It seems to me that if you configure the igate with the BME function active, when it reboot, if the BME is not connected, you get the message in the display, but also you cannot connect to the Igate anymore, until you reconnect the BME sensor.

SQ2CPA, 2024-03-05:
You can flash using VSC or you can flash using our installer but it is currently experimental. Both ways should work fine without any problems :) 

If we are talking about BME - I will test it after some time when I will modify the code related to it

closed 2024-03-07 by richonguzman