richonguzman/LoRa_APRS_iGate #39: Recommended Settings for Pinpoint APRS

travnewmatic, 2024-03-05:
First of all, THANK YOU SO MUCH for this amazing project.  Myself and a few others in our local amateur radio club have started investigating the capabilities of some LoRa boards running your software (iGate and tracker).  And we have already achieved some success!

Members of our group are accustomed to using Pinpoint APRS on the desktop.  In the release notes, I saw mention of Pinpoint APRS, and in the code it looks like I need to [uncomment a line](https://github.com/richonguzman/LoRa_APRS_iGate/blob/8663640070ea62e5451fd9ec87a9a2ed2fcf9c01/src/aprs_is_utils.h#L8), which I have done:
[image]

Did another build and upload, and got the board running.  I opened PuTTY and connected to the COM port the board was on, and got one of my trackers to beacon.  This was what I saw on the console:
[image]

Is this what I should expect to see after uncommenting that line?

I had not checked console output before i made the change to that line, so I'm not sure if the behavior has changed or not.  But at least there was some activity on the console when the iGate heard something.

On the Pinpoint APRS side, other than making sure the COM port and Serial Speed are set correctly, what should those settings be?
[image]

So far, when I 'Connect TNC' in Pinpoint APRS, the board seems to freeze.  No new beacons are acknowledged, and nothing changes on the T-Beam OLED display.  When I tell Pinpoint APRS to disconnect, the board appears to reboot, then gets to its 'running' state, and receives beacons properly (and passes traffic to aprs-is).

I think I'm close!  Any help would be appreciated, and please let me know what other information I can provide.

Thanks and have a great day!

SQ2CPA, 2024-03-05:
Hello, as a side note, I would like to add that in a while we will support TNC KISS via TCP/IP so you will be able to connect via APRSIS32 or Pinpoint. Of course we will include all screens with correct configuration for each program

travnewmatic, 2024-03-05:
Hello!

I am KJ5DCL from Richardson, Texas, USA!

Thank you so much for the lightning fastresponse!  I have done two things:
- I've reinstalled the iGate software using the .bat in the corresponding zip release for my particular board
- I've configured Pinpoint APRS like [this screenshot](https://github.com/richonguzman/LoRa_APRS_iGate/blob/main/images/PinPoint_01_configuration.png) (thank you for sharing that link)

And it works!
[image]
[image]

I know the guys I'm working with will be happy to know that they can use their existing APRS tools with your software.  Thanks again and we'll be in touch!

- Travis KJ5DCL

closed 2024-03-05 by travnewmatic