richonguzman/LoRa_APRS_iGate #45: Virus detected in Source files

N1obu, 2024-03-10:
I tried to download the latest source.zip and the source.tar.gz.  Both files detected a trojan virus in the file(s).

richonguzman, 2024-03-10:
hi,

The .zip file is generated automatic by Github with our firmware code but it could happened that in Windows OS a Trojan Warning could appear in the Windows Defender app. This is a sadly usual problem in Github with Windows and installers. Don't worry as all files have been tested with https://www.virustotal.com/gui/home/upload and passes without any problem.

SQ2CPA, 2024-03-10:
Can you tell exactly what file/files causes the problem after disabling AV for a moment and unpacking this zip? 

We know about it, most likely it is caused by the installer. Of course there is no virus, it is a false positive. This is a very tiring problem :(

N1obu, 2024-03-10:
Issue fixed on 3/10/24 download

closed 2024-03-10 by N1obu