richonguzman/LoRa_APRS_iGate #46: OTA upgrade procedure

iw2ejh, 2024-03-11:
I have a question related to OTA iGate upgrade:
in the webgui we can load "Firmware" and "LittleFS/SPIFFS"
if I want to upgrade the iGate I think I need the Firmware.
Can I get it directly from the zip file "ttgo-lora-v21.zip" (in my case) i downloaded from the link https://github.com/richonguzman/LoRa_APRS_iGate/releases/tag/V1.0.1 , extract the file called firmware.bin and uplade it?  then extract the file spiffs.bin and load it too ?  is this the correct procedure for an OTA upgrade?  Thanks

ClassAxion, 2024-03-13:
Hello. Flashing `firmware.bin` via OTA should work. I'm not sure about `SPIFFS`. In a while we will have dedicated OTA bins in releases packs.

SQ2CPA, 2024-03-28:
Fixed in https://github.com/richonguzman/LoRa_APRS_iGate/pull/64
For now you have OTA update binary in installer zip, SPIFFS you need to create yourself because you shouldn't flash clean SPIFFS because this will disconnect your device from WiFi

closed 2024-03-28 by richonguzman