richonguzman/LoRa_APRS_iGate #49: Web: Removing WiFi-APs ruins whole config

formtapez, 2024-03-16:
Hi, when you use the Web-Interface to remove all WiFi-APs and write the config, the configuration file gets corrupted and the Web-Interface is no longer usable (Error: "Failed to load configuration").

I think the error could be here (when the number is zero):
https://github.com/richonguzman/LoRa_APRS_iGate/blob/main/src/web_utils.cpp#L62

SQ2CPA, 2024-03-18:
Hello. When you get "Failed to load configuration" can you go to `http://YOUR_LORA_IP/configuration.json` and send the config? Of course please remove your passwords first! You can send it here in the issue or to my e-mail: `sq2cpa@gmail.com`

SQ2CPA, 2024-03-21:
Ok, this problem will be fixed in next update. If you want, you can go to this repo and test it now, release is available: https://github.com/SQ2CPA/LoRa_APRS_iGate

SQ2CPA, 2024-03-22:
Fixed in https://github.com/richonguzman/LoRa_APRS_iGate/pull/56

closed 2024-03-22 by richonguzman

formtapez, 2024-03-22:
I´m a little late, but i have tested it now.
The fix works. Thank you very much!