richonguzman/LoRa_APRS_iGate #51: Beacon path issue

jclifford73, 2024-03-17:
Hello Ricardo.
Firstly many thanks for your brilliant software.   I don't know how you do it!!!

I have a couple of issues.  The first one is that removing the beacon path in igate_config.json stops beacons being sent to APRS-IS, resulting in no icon being displayed on aprs-fi.  This is the line:

"path": "",

A configuration with no path should be allowed so beacons are not digipeated.

73 John GW4BVE

SQ2CPA, 2024-03-18:
Hello. Thanks for your issue. I will check that because you are right, empty path is allowed and correct according to the protocol. 73! Damian SQ2CPA

SQ2CPA, 2024-03-22:
Hello. This will be fixed in next update. If you want, you can go to this repo and test it, release is available: https://github.com/SQ2CPA/LoRa_APRS_iGate

SQ2CPA, 2024-03-22:
Fixed in https://github.com/richonguzman/LoRa_APRS_iGate/pull/56

closed 2024-03-22 by richonguzman