richonguzman/LoRa_APRS_iGate #54: Measure external volatage is greyed out in WebUI

S57PNX, 2024-03-19:
Hi all,

After upgrading to WebUI 1.0.0 from pre-UI firmwares, the "Measure external voltage" switch is grayed out and so is box for External Voltage Pin number. After enabling and disabling "Send Battery Telemetry" switch I am able to switch on the "Measure External Voltage" switch, but this doesn't persist across device restart and doesn't work. Even if I backup the config to json, change externalVoltageMeasurement to true and restore it to the device, it comes back as disabled after reboot and it doesn't perform the measurement.

Is anybody else seeing the same behaviour?

richonguzman, 2024-03-19:
Hi, this is a known bug and this will be taken care in soon (really) to be released version, 

sorry for the bug.

SQ2CPA, 2024-03-19:
Hello. This will be fixed in next update. If you want, you can go to this repo and test it, release is available: https://github.com/SQ2CPA/LoRa_APRS_iGate

SQ2CPA, 2024-03-19:
Yes you are right. I will check that.

Probably you missed one big update that removed stationMode and changed the config.

SQ2CPA, 2024-03-19:
Please check again, newest version is available at the same place. 
So from now "Send Battery Telemetry" and "Measure External Voltage" will work as separate options as before.

S57PNX, 2024-03-19:
Thanks, Send Battery Telemetry and Measure External Voltage work independently now.

I don't think I missed any firmware updates, both my test units were on the latest CA2RXU firmware which converted the old config to new one. From there I moved them both to SQ2CPA test firmware and had config issues. 

I will try to replicate with a fresh TTGO module tomorrow.

S57PNX, 2024-03-20:
I was able to replicate the problem when moving to the test firmware. 

I started with a TTGO LORA 2.1 module, used vstudio to get latest CA2RXU source code, built it and installed it on TTGO together with filesystem with my config. The web interface came up correctly prepopulated (I just had to reenter WIDE1-1). Then I used OTA to update to SQ2CPA - which didn't reboot as expected. I pressed Save to trigger a reboot which caused SQ2CPA firmware to boot. Finally, refreshing the UI gives me "Failed to load configuration" and makes me reenter previously mentioned config entries. Even restoring an older config does not help, I need to reenter and save.

SQ2CPA, 2024-03-21:
Ok, yesterday I also was able to replicate the problem so I will fix this today and I will let you know

S57PNX, 2024-03-22:
Unfortunately, even with the latest SQ2CPA firmware I get "Failed to load configuration" after the upgrade from latest CA2RXU until I reenter most of the config and save it.

SQ2CPA, 2024-03-22:
Ok, when you get "Failed to load configuration" can you go to http://YOUR_LORA_IP/configuration.json and send the config? Of course please remove your passwords first! You can send it here in the issue or to my e-mail: sq2cpa@gmail.com This way I will be able to diagnose better. Thank you for your patience!

SQ2CPA, 2024-03-22:
"Measure external volatage is greyed out in WebUI" Fixed in https://github.com/richonguzman/LoRa_APRS_iGate/pull/56
"Failed to load configuration" Fixed in https://github.com/richonguzman/LoRa_APRS_iGate/pull/57

closed 2024-03-22 by richonguzman