richonguzman/LoRa_APRS_iGate #55: Feature Request:  KISS output

N1obu, 2024-03-19:
Would it be possible to make a configurable option to all the USB interface output and allow input KISS data?

My hope is to allow the node to connect to an APRS client and act as a KISS TNC device and allow the client software do all the work

SQ2CPA, 2024-03-19:
Hello. In the next update will be TNC KISS server (TCPIP) and Serial KISS so this will work without any problems with APRSIS32, aprsdroid, etc. For now you can go here and test it if you want, release is available: https://github.com/SQ2CPA/LoRa_APRS_iGate

richonguzman, 2024-03-20:
> Would it be possible to make a configurable option to all the USB interface output and allow input KISS data?
> 
> My hope is to allow the node to connect to an APRS client and act as a KISS TNC device and allow the client software do all the work

please let us know how it went with the tests ;)

N1obu, 2024-03-20:
I will test in the next few days. Thanks for the info.

N1obu, 2024-03-20:
I compiled & uploaded the Dev code you provided.

I'm running Xastir and my TBeam attached as a Serial KISS TNC without any problems.

Tracker TBeam is heard by the i-Gate and passes the beacon to IS-APRS and Xastir.

Xastir beacon is displayed on the i-Gate and on  the Tracker.  However,  the Xastir beacon wasn't passed to IS-APRS.

I left it running and will check on it tomorrow to see if local (filtered m/100) traffic is passed from IS-APRS to RF.

SQ2CPA, 2024-03-21:
Great! Thank you very much for testing

closed 2024-04-16 by richonguzman