richonguzman/LoRa_APRS_iGate #59: Feature request: Turn wifi off after 10 minutes to save battery in Digimode

jlnvdr, 2024-03-24:
Turning WIFI off in Digimode after 10 minutes to save battery on Solar powered IGates in Digi-Mode. 
After powering off and on again WIFI should turn on for 10 minutes again.

Thx
Julian

richonguzman, 2024-03-24:
nice idea, maybe we should add a configuration issue to enable or not wifi in remote digirepeaters stations.

SQ2CPA, 2024-03-25:
Hello. For now, if you are using WiFi AP it will turn off after some time (defined in web ui) you mean something similar but for WiFi client?

jlnvdr, 2024-03-25:
> Hello. For now, if you are using WiFi AP it will turn off after some time (defined in web ui) you mean something similar but for WiFi client?

I know that, but the question remains the same is the power consumption with that gone?

SQ2CPA, 2024-03-25:
It should, I haven't tested it on the power supply - I can do that when I will have some time

richonguzman, 2026-09-30:
Digimode without WiFi has been added for a while

closed 2026-09-30 by richonguzman