richonguzman/LoRa_APRS_iGate #60: heltec_wireless_stick_lite envelop in platform.ini

HA7CQ, 2024-03-24:
The HTIT-WSL is composed up of an MCU (ESP32-S3FN8)

Defined in platform ini: board_build.mcu = esp32c3 is wrong

SQ2CPA, 2024-03-25:
Hello. Ok, sorry, one thing: there is name error - heltec_wireless_stick_lite is for heltec ht-ct62 module, not for heltec wireless stick board. I will change that name in next update so as not to cause confusion. 73! Damian SQ2CPA

HA7CQ, 2024-03-25:
Thanks! 
I tested the wireless stick lite, and work well with the heltec wireless stick envelop.  I think that the lite panel deserves a firmware without display management, they work perfectly in repeaters. We never look at the display on the mountaintop in the tower, it is a cheaper and more reliable panel. They are more energie efficient.  
Thanks Damian!

SQ2CPA, 2024-03-28:
Fixed in https://github.com/richonguzman/LoRa_APRS_iGate/pull/64