richonguzman/LoRa_APRS_iGate #61: Feature Request: GPS in iGate

N1obu, 2024-03-24:
It would be nice for the iGate to use the GPS instead of a static position.

richonguzman, 2024-03-24:
I have this on my TODO list for using a special button , if gps available, to get the current GPS position as the current igate/digirepeater GPS position

also im not very convinced of having a continuos moving igate station.

N1obu, 2024-03-24:
The intent of having the iGate with GPS is to set up a base-camp scenario with or without Internet capability and then deploying the trackers in that local area (almost like a hiker & camp).   This way we don't have to re-upload the 'json' config file with an updated position.

DC4FRT, 2024-03-27:
i would also assist the request, to have a mobile or portabel i-Gate if you use the iGate only for some hours or days on the same possition. maybe this could be an choosable option then on the webpage (a software button), i would not need an additional HW-Button
73 and all the best Thomas, DC4FRT

mytechguyri, 2024-03-27:
Even better if it could connect to a GPS via gpsd  I run a single gpsd server that serves gps coordinates to my entire network.

richonguzman, 2024-03-28:
can you explain me more about this gaps service or process? write me at telegram @richonguzman

> Even better if it could connect to a GPS via gpsd I run a single gpsd server that serves gps coordinates to my entire network.

iw2ejh, 2024-04-10:
May be using smartphone GPS to communicate the position to the "mobile iGate" could be a solution? using the web interface for this, and having a software switch to memorize the position is we put the iGate in a new fixed position? This will also help in deploying iGates in various locations.

richonguzman, 2024-10-21:
GPS in iGate is added in Release V2.1

closed 2024-10-21 by richonguzman