richonguzman/LoRa_APRS_iGate #62: Unstable Wifi ?

HansBaechle, 2024-03-27:
Hi Ricardo,
in the new Version, i've got some Wifi-Problems. it is not stable
on Version 2024.03.24. I have no Problems with Version 2024.03.18.
Also no Changes down with my WIFI-Network.
Here is the output from the device ttgo lora v2.1

- Sending Beacon to APRSIS --
-- Sending Beacon to APRSIS --
-- Sending Beacon to APRSIS --
[43213781][E][WiFiClient.cpp:422] write(): fail on fd 48, errno: 104, "Connection reset by peer"
Connecting to APRS-IS ...     Connected!
                                        (Server: dl.aprs2.net / Port: 14580)
-- Sending Beacon to APRSIS --
-- Sending Beacon to APRSIS --
-- Sending Beacon to APRSIS --
-- Sending Beacon to APRSIS --
-- Sending Beacon to APRSIS --
[47714267][E][WiFiClient.cpp:422] write(): fail on fd 48, errno: 113, "Software caused connection abort"
Connecting to APRS-IS ...     Connected!
                                        (Server: dl.aprs2.net / Port: 14580)
-- Sending Beacon to APRSIS --
-- Sending Beacon to APRSIS --
-- Sending Beacon to APRSIS --
[50414514][E][WiFiClient.cpp:422] write(): fail on fd 48, errno: 104, "Connection reset by peer"
Connecting to APRS-IS ...     Connected!
                                        (Server: dl.aprs2.net / Port: 14580)
-- Sending Beacon to APRSIS --
-- Sending Beacon to APRSIS --
-- Sending Beacon to APRSIS --
[53114780][E][WiFiClient.cpp:422] write(): fail on fd 48, errno: 104, "Connection reset by peer"
Connecting to APRS-IS ...     Connected!
                                        (Server: dl.aprs2.net / Port: 14580)
-- Sending Beacon to APRSIS --
-- Sending Beacon to APRSIS --
-- Sending Beacon to APRSIS --
[55815053][E][WiFiClient.cpp:422] write(): fail on fd 48, errno: 104, "Connection reset by peer"
Connecting to APRS-IS ...     Connected!
                                        (Server: dl.aprs2.net / Port: 14580)
-- Sending Beacon to APRSIS --
-- Sending Beacon to APRSIS --
-- Sending Beacon to APRSIS --
[58515340][E][WiFiClient.cpp:422] write(): fail on fd 48, errno: 104, "Connection reset by peer"
Connecting to APRS-IS ...     Connected!
                                        (Server: dl.aprs2.net / Port: 14580)
-- Sending Beacon to APRSIS --
-- Sending Beacon to APRSIS --
-- Sending Beacon to APRSIS --
[61215629][E][WiFiClient.cpp:422] write(): fail on fd 48, errno: 104, "Connection reset by peer"
Connecting to APRS-IS ...     Connected!
                                        (Server: dl.aprs2.net / Port: 14580)
-- Sending Beacon to APRSIS --
-- Sending Beacon to APRSIS --

Can I change something in my configuration? (I'm the Problem ?) :-)

73 from South Germany
Hans, DL7GAI

richonguzman, 2024-03-27:
hallo,

this is news to me... will check about it, but the wifi process has not changed at all.

S57PNX, 2024-03-28:
"Connection reset by peer" means that the remote side, dl.aprs2.net, sent a reset packet. So either dl.aprs2.net is actively rejecting or cancelling your connection, or a device in between (like your router or firewall) is doing it. 

Try to use a different aprs-is server for comparison, or simply use euro.aprs2.net for round-robin connection to different EU servers.

HansBaechle, 2024-03-28:
Hi Ricardo,
i think we can close this issue.
yesterday evening, i upgarded to Version 2024-03-27.
Now, about 17 hours later, no WiFi issue appeared.
so I Think it is solved
73
Hans, DL7GAI

closed 2024-03-28 by HansBaechle

HansBaechle, 2024-03-28:
Thank you very much