richonguzman/LoRa_APRS_iGate #63: Not sure if its my device or not

mytechguyri, 2024-03-27:
I just installed on my Heltec LoRa 32 Wifi V3
Everything seems to be functioning except for the LoRa radio.... I'm not sure if its my device, or something with the software.

I'm seeing the following activity from the serial terminal and the white "transmit" LED lights on the device... 
Connecting to APRS-IS ...     Connected!
(Server: aprs-is.wa1okb.radio / Port: 14580)
-- Sending Beacon to APRSIS --
---> LoRa Packet Tx    : WA1OKB-15>APLRG1,WIDE1-1:=4141.17NL07123.96W#LoRa APRS iGate https://github.com/richonguzman/LoRa_APRS_iGate
-- Sending Beacon to APRSIS --
---> LoRa Packet Tx    : WA1OKB-15>APLRG1,WIDE1-1:=4141.17NL07123.96W#LoRa APRS iGate https://github.com/richonguzman/LoRa_APRS_iGate
-- Sending Beacon to APRSIS --
---> LoRa Packet Tx    : WA1OKB-15>APLRG1,WIDE1-1:=4141.17NL07123.96W#LoRa APRS iGate https://github.com/richonguzman/LoRa_APRS_iGate

but I've got my HT programmed to 433.775 sitting right next to it, and I'm not hearing any activity on the frequency when I tell it to transmit a beacon.

richonguzman, 2024-03-27:
By HT you mean handy listening on UHF audio?

If this is the case , Lora packets only can be heard without using any squelch and very low (0.1w but powerful enough for other Lora units to be heard)

UHF is not the same as LoRa (and viceversa) although using same freqs

mytechguyri, 2024-03-27:
Yes, I've completely turned off squelch... and the HT is literally mere inches from the LoRa board, so even at the very low power, I expected I'd at least hear something.... but I'm getting absolutely nothing at all, dead silence...  This is my first go into LoRa, so perhaps I'm just mistaken about my expectations... Guess I'll have to buy a second device and see if they can be heard by each other.

richonguzman, 2024-03-28:
do you have another LoRa unit near by? or a friends LoRa iGate in the city?

S57PNX, 2024-03-28:
The serial log shows that your igate is beaconing (=transmitting its own position) to APRS-IS. I don't see any evidence that your igate is beaconing into RF - did you enable the "Send beacon to RF" switch and possibly the "Enable Lora TX" switch a bit lower on screen?

Also keep in mind that Lora modulation will be hard/impossible to hear on a FM radio, unlike the traditional AFSK APRS that made characteristic loud sounds on a FM radio. Certainly it will be hard to recognize even if you do hear something when the squelch is off.

ClassAxion, 2024-04-03:
@mytechguyri has the problem been resolved?

SQ2CPA, 2024-04-04:
@mytechguyri has the problem been resolved?

closed 2024-06-18 by richonguzman