richonguzman/LoRa_APRS_iGate #66: NEW feature. E-ink

Szetya, 2024-03-31:
First test. 😉 
https://youtu.be/5zWcB2fp9pM?si=NeCK4KrWzQeXxW-O

SQ2CPA, 2024-04-01:
Ok but why, for what purpose? My IGate/Digi stations are automatic stations and I don't look at them, this feature is better for tracker. I can see the statistics etc. via website or other way on my TV screen. 

If you are connected to grid you don't need to save power and if you want to save power, go low power mode etc. then you don't want to use any screen.

This is just my opinion about this feature.

Szetya, 2024-04-01:
Well. For what purpose? 
Maybe just experiment, curiosity, interest, to explore the possibilities.
Not an option for Tracker because it is slow due to frequent screen refreshes.
But for Igate it is suitable and especially because I only update the information on the display when there is a change. This also speeds up the time to run the code once.
I am aware that you don't watch this all the time and there are other options (web, etc). 
But then I could ask the question, why is the oled display implemented in the code? ;)
Maybe as you saw it is not an accidental choice of HWSL since it has no display. Precisely because I find the display unnecessary for igate. And here I go back to my first sentence: curiosity, experiment...
(You know that this is not my first attempt. Just a few: tracker T-deck, Helteck wireless tracker, lilygo T-QT pro, dualscreen oled.)
[image]
[image]
[image]