richonguzman/LoRa_APRS_iGate #72: version 2024.04.09 config issue

iw2ejh, 2024-04-11:
Hello,
I updated to latest version "2024.04.09" and noticed a couple of issue in the configuration.
I updated building and loading the firmware using VSC, then loaded the filesystem.
I had to correct from the Access Point some parameters. I noticed the following problems:
- the beacon is duplicated,  _xxxx become _xxxx_xxxx  
- the display never shut down selecting the relative function.
- I also noticed that you cannot select the Digipeater icon if you have a BME connected.
Tnx
iw2ejh

richonguzman, 2024-04-11:
hi

please download not again :
duplicated comment fixed when Wx is selected

will check about the display

when Wx data is selected, the symbol/icon should be the blue Circle + L, or aprs.fi won't process it as wx data and only as comment

SQ2CPA, 2024-04-12:
After selecting BME sensor your symbol will be override to `L_` after saving config. That because any other symbol from the list doesn't make sense because your WX data will be inserted as comment and won't be parsed as WX data with any other symbol than `L_`

iw2ejh, 2024-04-12:
> After selecting BME sensor your symbol will be override to `L_` after saving config. That because any other symbol from the list doesn't make sense because your WX data will be inserted as comment and won't be parsed as WX data with any other symbol than `L_`

Do you mean the blue circle symbol? or the L overlay?  anyway what if we send the digi symbol with the beacon and the blue circle symbol only when the meteo data is sent? is this possible, so we could know what kind of mode the LoRa device is configured for?  Usually we can configure different kind of beacon with APRS devices. Thanks

N1obu, 2024-04-13:
I had a problem configuring.  I uploaded new image to my TBeam v1.1 and it stayed on the initial screen.  I could not upload config file.  No access-point was available for me to configure.  I had to erase flash and downgrade back to version 2024.03.28.

richonguzman, 2024-04-13:
You need to try again to upload filesystem or it won't work

N1obu, 2024-04-13:
I did several times.  I even completely erased the TBeam and also cleaned the build.  When I downgraded and performed the same load procedure, everything worked fine.  I am using a TBeam v1.1 (915Mhz).  You can see my nodes online at aprs.fi. Search for N1OBU-11 (igate) and -12 (tracker).

richonguzman, 2024-04-16:
how did you erase the T-beam?

a way to erase all is to write igate firmware (firmware and also filesystem) or meshtastic and go back to write Tracker Firmware and filesystem

sometimes when adding new stuff the .json file of the configuration (although having the same name) won't work and needs to be erased from internal memory

SQ2CPA, 2024-04-16:
Can you check latest release? We have fixed some problems releated to config. 73! Damian SQ2CPA

N1obu, 2024-04-16:
I tried then new code. Same results. The code loads, the TBeam resets, I load the file-system (config file). The TBeam resets again, but stays on the initial boot screen. I have tried two different TBeams (one 915Mhz and one 433Mhz), same results.

Using the same loading procedure and using software build 2024.03.28. I am able to load my TBeams.

I am using Visual Studio with PlatformIO.

richonguzman, 2024-04-17:
wait a day for a new way of installing and avoid VSCODE completely ;)

N1obu, 2024-04-18:
Here is some more troubleshooting info.  I did a fresh build and upload.  I just changed my callsign in the config json file.  I then watched putty to see what the serial output would be..  Here is my output:

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13192
load:0x40080400,len:3028
entry 0x400805e4
AXP192 init done!

Starting Station: N1OBU-13   Version: 2024.04.13
Starting LoRa failed!

==============
As stated before.  If I downgrade to _2024.03.28_ build, everything works fine.

richonguzman, 2024-04-18:
did you change the frequency on the tracker_conf.json from 433 to 915ish ??

N1obu, 2024-04-18:
Yes. I'm using 915Mhz TBeams. I switched to my 433Mhz and it works.  

However, the 2024.03.28 build works with my 915Mhz TBeams (igate and tracker).

I can't tell which TBeam has what SX chip on it. I tried searching my past orders through Aliexpress, but no technical information is given. 

All I can tell, some have the NEO-6M GPS and some have the NEO-M8N GPS.  All of them are T22_v1.1.

S57PNX, 2024-04-18:
Can you attach a photo of your T-beam board? At least the frequency band should be written on the LORA chip, on newer generations of the SX chip even the model is written.

richonguzman, 2024-04-18:
> Yes. I'm using 915Mhz TBeams. I switched to my 433Mhz and it works.
>
> However, the 2024.03.28 build works with my 915Mhz TBeams (igate and tracker).
>
> I can't tell which TBeam has what SX chip on it. [...]

I need to validate that you switched the frequency on the igate_conf.json and uploaded over VSCODE.

(as this is the only way right now to use those freqs)

N1obu, 2024-04-18:
Yes. I changed the frequency in the JSON config file to 915775000. It worked on the 3.28 build. However I used the TBeam with the NEO-6M GPS.

For testing, I used my spare TBeam with the NEO-M8N GPS.

I found out the 915Mhz TBeams with the 6M GPS have the SX1276 chip. 915Mhz TBeams with M8N GPS have the SX1262 chip.

The 433Mhz TBeam with the 6M GPS have the SX1278 chip.

I'm wondering if the reason 915Mhz worked is because the SX1276 and SX1278 chips have the same architecture. And the SX1262 has a different architecture.

Here are the pictures you asked for
[image]
[image]

richonguzman, 2024-04-19:
lets start again:

you have a T-beam V1.1 with SX1262 or SX1268 module?  

also, was the chip different because sx1262 is 433 and yours could be sx1268  for 915 ?

N1obu, 2024-04-19:
I have several TBeams. All are version 1.1, but they have different LoRa chips based on the GPS chip.

SX1276 is 915Mhz (GPS 6M)
SX1262 is 915Mhz (GPS M8N)

SX1278 is 433Mhz (GPS 6M)

Here is my reference:

[image]
[image]

richonguzman, 2024-04-19:
all worked with last firmware? and now only a few work?

richonguzman, 2024-06-18:
will add 915 for all boards with SX1276 very soon

closed 2024-06-18 by richonguzman