richonguzman/LoRa_APRS_iGate #75: OTA username/pw is not read from config

S57PNX, 2024-04-14:
For a while I had problems accessing the OTA page - if I had a password configured, it wasn't working. If I tried to remove the username/password, I was still being asked for an username/password for OTA. 

Eventually I added a couple of serial.println calls to ota_utils.cpp and found that regardless of the configured OTA username/pasword in the config file, the username and password are always literal string "null". This is with various firmwares since February.

How to replicate: start with a virgin board, upload firmware and default filesystem (without OTA username/password) to it. Connect to AP,  Click on "Update OTA" and when asked for username/password, enter null/null. 

Anybody else seeing this?

Mane76, 2024-04-14:
Yes, confirmed and already adressed to Ricardo

SQ2CPA, 2024-04-14:
Yes, we found the problem and this will be fixed within 24 hours

SQ2CPA, 2024-04-16:
Ok please check the fresh release. Now it should work :) 73! Damian SQ2CPA

S57PNX, 2024-04-17:
Confirm it is fixed for me, thank you!

richonguzman, 2024-04-17:
great!

closed 2024-04-17 by richonguzman