richonguzman/LoRa_APRS_iGate #77: 2024-04-13 version

iw2ejh, 2024-04-19:
today I updated my Solar test digi +WX and noticed a few things:
- in the digi+WX, if the BME is disconnected the digi/iGate do not start and display "error, BME configured but BME missing" would it be possible to bypass this, and have the digi working even in case the BME lost connectivity or get damaged?
- the display is not shutting down  even if selected.
- I enabled "power saving mode", and system "shut down at low level:". When I write the level, how the parameter is written? with coma, or point? with decimal? I put 3.2  with a point and one decimal but not sure it is the right format.
- The wiki warm about carefull connection to monitor external battery, and indicate to check for a specific information. But in fact this specific information is missing (or I couldn't found)
Thanks for the great job
iw2ejh

richonguzman, 2024-04-19:
1) good idea...  will check it out
2) will check it out
3) with "." always
4) where does it say specific information? do you mean about voltage divider?

iw2ejh, 2024-04-19:
#"where does it say specific information? do you mean about voltage divider?"
   - In the wiki,   https://github.com/richonguzman/LoRa_APRS_iGate/wiki/11.-VSCODE-iGate-Configuration   
   -OTHER
  - "externalVoltageMeasurement" : true will measure a 12V battery through a voltage divider resistor net and send its Voltage to APRS-IS. Default is false. **_Look for special info on this Wiki to connect the battery and voltage divider to avoid burning the board._**    

-but I cannot found other special info.

richonguzman, 2024-04-19:
between all the changes this was moved and deleted...  will add it soon

meanwhile:

 its a voltage divider for external voltage measurement. 
 Max Voltage to read 15v
 resistor 1: 100K (between + of battery to ESP32 pin)
 resistor 2: 27K from ESP32 pin to GND

oe5reo, 2024-04-23:
In "Received Packets" the last 20 received packets should be listed, but here on my end it only shows 11 ... not a big deal, but wanted to let you know.

73 Martin
OE5REO

richonguzman, 2024-06-18:
many things of the list are already done... I have taken notes of the missing ones to add/Check them soon

closed 2024-06-18 by richonguzman