richonguzman/LoRa_APRS_iGate #81: Feature request : comment text in syslog output

ON3JT, 2024-04-27:
Is it possible to add the comment text also in the syslog output?

Example (raw data from APRS.fi) : 2024-04-27 08:08:47 CEST: SP7TBS-7>APLOX1,WIDE1-1,qAO,ON2OT-11:!/4WFoOY-=u5jX**Lora Tracker 434.855 @1k2 QSO 145.500FM**

So that also **Lora Tracker 434.855 @1k2 QSO 145.500FM** is in the output string.

Keep up the great work!

richonguzman, 2024-04-27:
Right now is cutting out the comment?

Sure , I'll fix it

ON3JT, 2024-04-28:
Thanks in advance.

This is a raw string taken from the syslog output : - ON2OT-11 CA2RXU_LoRa_iGate_1.3 - - - Rx / GPS / ON3JT-7 / APLRT1 / WIDE1-1 / -61dBm / 9.50dB / -1868Hz / 51.17125N / 4.46718E / 0.0km

And this is the raw in aprs.fi : 
2024-04-27 22:02:17 CEST: ON3JT-7>APLRT1,WIDE1-1,qAO,ON2OT-10:!/4Z-)OYbt[!!QTesting tracker

It would be awesome if the Testing tracker comment can be included in the syslog output. This gives probably a little nightmare in the choice of what to use as a delimiter.

richonguzman, 2024-04-28:
should we add this to the very end?

ON3JT, 2024-04-28:
> should we add this to the very end?

That is ok. But I'm thinking ahead, what if someone uses the forward slash in the comment? Using the tilde instead ~ ?

closed 2024-08-13 by richonguzman