richonguzman/LoRa_APRS_iGate #82: Compressed location details

ON3JT, 2024-04-27:
I noticed on the TTGO screen and in the syslog output that compressed data results in a longitude and latitude of 0.00N, 0.00E.
The raw data is sent to the APRS IS servers without any problem, they handle it from there :-)

But is seems on screen that some stations are 5700 km away, while in reality it is just 4km.
I had the problem too, so I attached a bit of ugly code :
[APRS_compressed.txt](https://github.com/richonguzman/LoRa_APRS_iGate/files/15137345/APRS_compressed.txt)

I test first if it is compressed data, if so, then proceed to correct function.

ON3JT, 2024-04-27:
I'll need to clarify it a bit. Using the tracker that also sends compressed data I didn't noticed the wrong location data. Some bug that it occures in rare situations?

richonguzman, 2024-04-28:
Hi

Could it be that the gps data inserted in webconfig page was 0.0, 0.0?

Or wrong values with "," instead of "."

Some send data with 0.0 , 0.0 to avoid being seen

ON3JT, 2024-04-28:
I added a file with details. I hope that it help to explain it a bit :-)

[differences.txt](https://github.com/richonguzman/LoRa_APRS_iGate/files/15142338/differences.txt)

richonguzman, 2024-04-28:
> I added a file with details. I hope that it help to explain it a bit :-)
> 
> [differences.txt](https://github.com/richonguzman/LoRa_APRS_iGate/files/15142338/differences.txt)

so, I should add the comment extracted from LoRaPacket and add it at the last part of the syslog log packet?

ON3JT, 2024-04-28:
> so, I should add the comment extracted from LoRaPacket and add it at the last part of the syslog log packet?
I believe this is for the other topic ;-)

richonguzman, 2024-05-20:
@ON3JT can you try "syslog-mod" branch which should deliver comment of compressed (base91) and not compressed gps data on Rx packets

let me know how it goes

richonguzman, 2024-06-18:
this seems to bee solved, so closing

closed 2024-06-18 by richonguzman