richonguzman/LoRa_APRS_iGate #83: PinPoint serial KISS only produces a callsign and not a callsign-SSID position report

AG7U, 2024-04-28:
I have four trackers:  WA0DE-T1, WA0DE-T2, WA0DE-T3, WA0DE-T4.  (WA0DE is our club callsign.)  When using Pinpoint serial KISS I only see WA0DE and not WA0DE-T1 or WA0DE-T4.  Without the associated SSID I cannot differentiate the trackers.  Are you outputting the SSID with the callsign for PinPoint serial KISS or just the callsign with no SSID?  Can the SSID be added to the callsign?  

In PinPoint you can view a “Comm Monitor Window”.  Below are two console outputs copied from the comm monitor.  
Serial KISS in PinPoint:
WA0DE>APLRT1,WIDE1-1:!/:OYU4!$v<PkQ Bat=4.18V (100%)
WA0DE>APLRT1,WIDE1-1:!/:OZj4!$`<P`Q Bat=4.16V ( 0mA)

APRS-IS in PinPoint:
WA0DE-T1>APLRT1,WIDE1-1,qAR,WA0DE-G1:!/:OZ'4!%D<PeQ Bat=4.18V (100%)
WA0DE-T4>APLRT1,WIDE1-1,qAR,WA0DE-G1:!/:OVd4!$l<PnQ Bat=4.16V ( 0mA)

richonguzman, 2024-04-28:
Can you repeat All test but using correct ssid for all?

WA0DE-T1 or WA0DE-G1 are not valid, as T1 or G1 are not valid SSID

USE 0-15 for testing:
Trackers from 6,7,8,9 and igate 5

It this works it has to do with how AX25 on some peograms do not work with ssid different 0-15

AG7U, 2024-04-28:
Thank you for your quick response to my issue.  You nailed it!  Works fine with SSID 0-15.  Any chance this behavior could be changed in a future release?  No rush.  For now I will continue to use a LoRa decoder developed by DJ7OO, which does display the special SSIDs in PinPoint.

richonguzman, 2024-04-29:
you mean changing the AX25 limit of bites it can work to have only 0-15 in ssid?

this is how it is performed the ax25. So no-one can changes this.

the solution would to use only text/serial output of the pc-tncs and avoid the ax.25 encapsulation

AG7U, 2024-05-02:
In iGate firmware version 2024.01.12 I can uncomment #define TextSerialOutputForApp in the aprs_is_utils.h file and that action produces the text/serial output I want to see for non-standard SSIDs.  Is there a way to bypass the need for Internet?  I am unable to start and sustain text/serial output without first connecting to WiFi and second maintaining a connection with the aprs server.  My objective is to deploy trackers and digis that will beacon to an iGate that provides text/serial output to PinPoint for bicycle and motorcycle races in areas where Internet does not exist.  

Did the text/serial output change in somewhere between version 2024.01.12 and 2024.04.23?  With 2024.01.12 I can see non-standard SSIDs in PinPoint.  In 2024.04.23 I cannot see non-standard SSIDs in PinPoint.  In 2024.04.023 there is no longer an option to uncomment #define TextSerialOutputForApp in the aprs_is_utils.h file.

richonguzman, 2024-05-02:
> In iGate firmware version 2024.01.12 I can uncomment #define TextSerialOutputForApp in the aprs_is_utils.h file and that action produces the text/serial output I want to see for non-standard SSIDs. [...]

now I get the problem....  I should see about fixing this soon

richonguzman, 2024-08-13:
Ssid T1, T2 and more are not encodable in ax25. Kiss protocol is ax25-packets-over-serial.

closed 2024-08-13 by richonguzman