richonguzman/LoRa_APRS_iGate #84: Display always on

IZ1HMT, 2024-04-29:
With the Heltec V3 card the display always remains on, even if the correct option and seconds are selected in the menu. This happens if the card does not connect to a wi-fi network or if it connects as an AP. However, if it is connected to an access point, it turns off correctly based on the seconds and the option selected in the menu.

richonguzman, 2024-04-29:
so you want to turn off when AP also?

IZ1HMT, 2024-04-29:
As I read in another thread it would be useful to have 3 options:
-Always on
-On only when receiving/transmitting or exchanging data 
-Always off

richonguzman, 2024-04-29:
right now there are 3 modes available:
-always on,
-few seconds on (time > 0)
-always off (time=0)

please check if time = 0 works for you (just tested here and screen does not turn on)

IZ1HMT, 2024-04-29:
Ok, but it only works when Wi-Fi is connected. If not connected in AP or Wi-Fi mode, the display always remains on regardless of the option I select.

HJ5JMS, 2024-05-07:
Hi, I have the same issue on TTGO 1.6, the display stays on even when there is an active connection to the AP.

closed 2024-06-18 by richonguzman