richonguzman/LoRa_APRS_iGate #86: Feature request: Note field in webgui

S57PNX, 2024-04-30:
This might be banal, but can we have a note field in GUI for internal notes about a gate? I would like to add my "friendly" name for the site and perhaps some other sysadmin details, just for me to see it next time I login to the webgui. Currently I need to always look up the call/SSID on the map because I am getting confused between all the gates/digis that I manage, to make sure I am working on the correct one. Thanks!

richonguzman, 2024-04-30:
Hi,

on the GUI is says right away the Callsign that you have on the igate... 

please explain if this is different from what I just wrote.

S57PNX, 2024-04-30:
One igate is S57PNX-12, one is S57PNX-14, one is S57PNX-13, one is S57PNX-8, one is S57PNX-15 etc. etc. I can reach all of them via network from my home. It's easy for me to get confused which one is which and I always double check the callsign by looking on aprs.fi map.

If you give me a little field for my notes, I can write the name of the location or street address or  maybe the phone number of the site owner or  any other information for my own notekeeping. Not for public display on the map.

Hope this explains it.

richonguzman, 2024-04-30:
I get now the request.... will see how to do this

S57PNX, 2024-06-26:
Works OK for me,  OK to close.
[image]

closed 2024-06-26 by richonguzman