richonguzman/LoRa_APRS_iGate #88: Wiki needs improvement for connecting BME280

mytechguyri, 2024-05-08:
The current wiki only says this about connecting a BME280

Adding BME280
BME280 (and BMP280) modules could be added by connecting them as I2C. Default I2C direction is 0x76 and some BME boards have pins that need to be joined to get this direction. Most of this boards don't need this.

To connect the BME280 to iGate Board use:

SDA to IO21
SCL to IO22
GND to GND (any)
VCC to 3.3V
To see this data on the Oled Screen or transmit it as Wx telemetry activate it on the igate_conf.json file.

These instructions are too vague... Connect SDA to IO21?   on what board?  I'm not even sure if the Heltec Wifi Lora 32 V3 even has the I2C bus pins exposed on the header based on this diagram, nor do i know what pins the iGate software is expecting it to be connected to.
[image]

mytechguyri, 2024-05-08:
I'm told others have had success with bme280 connected to Heltec Wifi Lora 32 V3 GPIO 41/42 over I2c with this code:

#define BME_SDA 42
#define BME_SCL 41

bool wireStatus = Wire1.begin(BME_SDA, BME_SCL);
status = bme.begin(0x76, &Wire1);

I looked through the code, and i don't see anywhere the pins are defined though.

richonguzman, 2024-05-10:
i will try to add a fix as HELTEC V3 need to use the Second I2C as BME module connector

stay tuned for testing FW soon

closed 2024-06-18 by richonguzman

lxndrJ, 2025-04-07:
Any updates still failing to connect BME280 to my V3.2 board?