richonguzman/LoRa_APRS_iGate #89: Beacon logic issue

jclifford73, 2024-05-09:
Hi Ricardo

Hardware: Heltec V3

"beacon": {
        
        "path": "",
        "sendViaAPRSIS": true,
        "sendViaRF": true

"aprs_is": {

        "active": true,
"lora": {
        
        "txActive": true,
        "rxActive": true

With these settings beacons are sent to both APRS-IS and RF correctly

If the "sendViaRF": true is changed to "sendViaRF": false  then no beacons are sent to RF or APRS-IS
If  "path": "", is then set to  "path": "WIDE1-1",  beacons to APRS-IS are restored

So path should not have to be set WIDE1-1 to allow beacons to be sent to APRS-IS.   A null path should be allowed.

Thanks for all your work on this great firmware.

73 John GW4BVE

richonguzman, 2024-05-09:
Hi John

sendViaRF:false  + path = "" = no beacons over APRS-IS  + no beacons over RF ??
sendViaRF:false + path = "WIDE1-1" beacons are sended normally (over APRS-IS) ?

PS: with sendViaRF:false + path = "WIDE1-1" are beacons sended normally over RF ?

jclifford73, 2024-05-13:
Hi Ricardo
Hardware: Heltec V3
Configuration changes made via the web interface

All tests:
Send our beacon to APRS-IS - True
Enable LoRa tx - True

Test 1:
Beacon path - WIDE1-1
Send beacon via RF - False
Result:
APRS-IS beacon OK
No RF beacon
A path of WIDE1-1 is required to send an APRS-IS beacon.  See test 2 below.

Test 2:
Beacon path - ""
Send beacon via RF - False
Result:
No APRS-IS beacon 
No RF beacon
In this test an APRS-IS beacon should be sent, but it is not sent.

Test 3
Beacon path - ""
Send beacon via RF - True
Result:
No APRS-IS beacon 
RF beacon OK
In this test the RF beacon is OK even with the "" path but the APRS-IS beacon is not sent

Test 4
Beacon path - "WIDE1-1"
Send beacon via RF - True
Result:
APRS-IS beacon OK
No RF beacon
In this test the RF beacon is not sent even though it is enabled

I hope this answers your question Ricardo.  What is your intended operation of the beacon logic?

73 John GW4BVE

richonguzman, 2024-10-07:
I will close this as Code has changed a lot last months so I will try to test all over again and again and again to be sure this all works ;)

closed 2024-10-07 by richonguzman