richonguzman/LoRa_APRS_iGate #90: voltage battery on Heltec V2

Dom83170, 2024-05-09:
Hello,

I can't have the voltage battery with the Heltec V2 is it normal or I made something wrong?

Congratulation for the job

Dominic

richonguzman, 2024-05-09:
> Hello,
> 
> I can't have the voltage battery with the Heltec V2 is it normal or I made something wrong?
> 
> Congratulation for the job
> 
> Dominic

hi!

please confirm "sendBatteryVoltage" is enabled on WebConfig UI

Dom83170, 2024-05-10:
>> Hello,
>> 
>> I can't have the voltage battery with the Heltec V2 is it normal or I 
>> made something wrong?
>> 
>> Congratulation for the job
>> 
>> Dominic
> 
> hi!
> 
> please confirm "sendBatteryVoltage" is enabled on WebConfig UI
> 
> [2].

Hello,

Yes, I confirm.

I had an other with a TTgo who runs well too.

Thanks for your help

Dominic

Links:
------
[1] 
https://github.com/richonguzman/LoRa_APRS_iGate/issues/90#issuecomment-2103095838
[2] 
https://github.com/
--=_86b166354a21179051fa60fb5bd1edd9
Content-Transfer-Encoding: quoted-printable
Content-Type: text/html; charset=UTF-8

<html><head><meta http-equiv=3D"Content-Type" content=3D"text/html; charset=
=3DUTF-8" /></head><body style=3D'font-size: 10pt; font-family: Verdana,Gen=
eva,sans-serif'>
<p id=3D"reply-intro">Le 09/05/2024 19:28, Ricardo Guzman (Richonguzman) a =
&eacute;crit :</p>
<blockquote type=3D"cite" style=3D"padding: 0 0.4em; border-left: #1010ff 2=
px solid; margin: 0">
<div id=3D"replybody1">
<p><br /></p>
<blockquote type=3D"cite" style=3D"padding: 0 0.4em; border-left: #1010ff 2=
px solid; margin: 0">
<p dir=3D"auto">Hello,</p>
<p dir=3D"auto">I can't have the voltage battery with the Heltec V2 is it n=
ormal or I made something wrong?</p>
<p dir=3D"auto">Congratulation for the job</p>
<p dir=3D"auto">Dominic</p>
</blockquote>
<p dir=3D"auto">hi!</p>
<p dir=3D"auto">please confirm "sendBatteryVoltage" is enabled on WebConfig=
 UI</p>
<p style=3D"font-size: small; -webkit-text-size-adjust: none; color: #666;"=
ichonguzman/LoRa_APRS_iGate/issues/90#issuecomment-2103095838" target=3D"_b=
lank" rel=3D"noopener noreferrer">view it on GitHub</a>, or <a href=3D"http=
AVCNFSM6AAAAABHPFOGZCVHI2DSMVQWIX3LMV43OSLTON2WKQ3PNVWWK3TUHMZDCMBTGA4TKOBT=
HA" target=3D"_blank" rel=3D"noopener noreferrer">unsubscribe</a>.<br />You=
 are receiving this because you authored the thread.<img src=3D"https://git=
hub.com/notifications/beacon/ATOWBKKULEB6I3AMED6H7KLZBOW3DA5CNFSM6AAAAABHPF=
OGZCWGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTT5LKZB4.gif=
" width=3D"1" height=3D"1" /><span style=3D"color: transparent; font-size: =
0; display: none; visibility: hidden; overflow: hidden; opacity: 0; width: =
0px; height: 0px; max-width: 0; max-height: 0; mso-hide: all;">Message ID: =
<span>&lt;richonguzman/LoRa_APRS_iGate/issues/90/2103095838</span><span>@</=
span><span>github</span><span>.</span><span>com&gt;</span></span></p>
</div>
</blockquote>
<p>Hello,</p>
<p>Yes, I confirm.</p>
<p>I had an other with a TTgo who runs well too.</p>
<p>Thanks for your help</p>
<p><br /></p>
<p>Dominic</p>

</body></html>

--=_86b166354a21179051fa60fb5bd1edd9--

richonguzman, 2024-05-10:
I would need for you to check for the specific version of your board here:

https://docs.heltec.org/en/node/esp32/wifi_lora_32/hardware_update_log.html#v2

as some pins get moved a lot

lemielek, 2024-05-19:
Is't it the same what is on Meshtastic that v2 has something bad with battery operations and external BMS is needed? And that is why V2 is not officially supportem with V2?

richonguzman, 2024-05-20:
there area few versions of Heltec V2 , so this maybe the reason why.

I've added the one someone sended me the values to validate it and it worked for him later when doing the mods.

I don't know what the Meshtastic guys tried with it.

If you send me a picture of the board we could start testing it

lemielek, 2024-05-20:
http://community.heltec.cn/t/heltec-wifi-lora-v2-battery-management/147/42
Via
https://www.reddit.com/r/meshtastic/comments/1b2q78r/heltec_v2/

I have only version for 868. Can make pic of it at about 4pm UTC.

richonguzman, 2024-05-20:
I've uploaded to branch "heltec v2 mod" the posible mods needed for you

heltec v2 has many versions and we are testing reading battery on pin37 as Heltec V2.1 

please test and confirm.

closed 2024-06-18 by richonguzman