richonguzman/LoRa_APRS_iGate #91: APRSdroid kiss tnc connection lost (TNC KISS Server)

ltenconi, 2024-05-09:
When using the TNC Server option with APRSdroid kiss tnc, APRSdroid gets disconnected and tries to reconnect. The TTGO is logging(Syslog) some packets from APRSdroid so the connection seems to be partially working.  
There are no specific details for APRSdroid in the TNC wiki section https://github.com/richonguzman/LoRa_APRS_iGate/wiki/14.-TNC-information, so I might be missing some configuration to make this work.

Excellent work Ricardo!

ltenconi, 2024-05-10:
Update: Seems to be working by disabling the socket time out option in Aprsdroid.

closed 2024-05-10 by ltenconi

ltenconi, 2024-05-10:
I can confirm TX is working with the following APRSdroid settings:

**Connection protocol:** TNC(KISS)
**Connection type:** TCP/IP
**Start up delay:** 300 ms
**Server:** igate_ip_address:8001
**Socket TCP time out: 0 -> DISABLED!**

richonguzman, 2024-05-10:
great info! will try to add this to a soon to be made TNC wiki