richonguzman/LoRa_APRS_iGate #92: Heltec V3 (esp32 s3) install failed - invalid header: 0xffffffff [Web flasher]

tanovich, 2024-05-09:
Hi, I got new Heltec v3 board and i wasn't able to make it work with iGate firmware. Web flasher prompts successful flashing, but OLED stays black. I checked serial monitor and theres looped message "invalid header: 0xffffffff". 

I also tried to use shell scripts provided in last release with same result. In order to make it work i needed to change esptool line by adding different chip type esp32-s3. 

To check if this is hardware related issue i flashed meshtastic (also web-flasher) and it worked. 

Am i missing something?

richonguzman, 2024-05-09:
hi,

there is a know bug with the installers we provided for this board ....  so the only way for this board (until a few days when we will upload fixes) is installing it from VSCODE

richonguzman, 2024-06-18:
HELTEC V3 is now added as a valid installer in new Web Installer v1.0.3

closed 2024-06-18 by richonguzman