richonguzman/LoRa_APRS_iGate #94: LoRa Chip issue

iw2ejh, 2024-05-15:
with my ttgo-lora32-v21 iGate, yesterday I lost power and battery went down.
When Power recovered, the iGate started to send to APRS-IS server a lot of time the same device position even if it do not heard it anymore.
I connected via the wifi and disabled the LoRa RX and the iGate stopped sending these ghost positions.  If I reenable the LoRa RX the problem restarted. I did a reset (push button) a power off power on but the problem still exist.
I then take down the iGate and flashed the latest version of the FW using VSC.
Now after the new flashed FW, the problem disapeare, but the LoRa chip now seems not receiving/decoding any signal, the other functions of the iGate is working as usual.
Is there a way to reset the LoRa chip or recover from this situation?
Thanks   
[image]

richonguzman, 2024-05-15:
hi, 

I've never heard of reseting the LORA module ..... could it be damaged with the power-off issue?

You just validate that the ESP32 works fine , and its a pure LoRa issue?

flashing both firmware and filesystem from VSCODE did solve the issue?
if not test MESHTASTIC on it and get back into LORA APRS firmware.

iw2ejh, 2024-05-15:
>>>You just validate that the ESP32 works fine , and its a pure LoRa issue?
Not sure if it is really a pure LoRa issue, if I disable LoRa RX in the config, the iGate stop to send hundred of data to the syslog server. 
>>>flashing both firmware and filesystem from VSCODE did solve the issue?
No, I flashed with VSCODE the latest version both firmware an  filesystem
I also flashed and erased everything and flashed "first flash or Factory reset" mode, got the same behavior
>>>if not test MESHTASTIC on it and get back into LORA APRS firmware.
I tried but was unable to flash using the web-flasher of meshtastic. I need to learn more about why it is not working.

I attach the log I got from the web-flasher and one from the syslog server.
[esp-web-tools-logs.txt](https://github.com/richonguzman/LoRa_APRS_iGate/files/15326338/esp-web-tools-logs.txt)
[192.168.1.72_20240515.txt](https://github.com/richonguzman/LoRa_APRS_iGate/files/15326344/192.168.1.72_20240515.txt)

iw2ejh, 2024-05-16:
Next step:
I was able to flash Meshtastic on the ttgo-lora32-v21, then I flashed again using the webflasher the APRS FW. then I connected to the internal AP and configured some of the parameters. I then reboot and tested, so far so good. Until I configured the low power mode. at that moment, the next reboot, the behavior reappeared.  I went immediately to the web interface and disabled the "low power mode" and the problem disapeared.   So it seems the problem have something to do with this setting.   I attach the log showing the debug log after the first firmware flash, some reboot due to setting updates, then the problem, and after when the problem disapeare  switching off the low power mode
[esp-web-tools-logs (2).txt](https://github.com/richonguzman/LoRa_APRS_iGate/files/15335313/esp-web-tools-logs.2.txt)

richonguzman, 2024-05-16:
ok, so this is only on the "low power mode" experimental part

this is meant for testing only on HT-CT62 module so please dont use it on another board for now.

at least you got it and now its working as it should ;)

iw2ejh, 2024-05-16:
I have this function active on another ttgo-lora32-v21 iGate I use with solar panel, without connection to internet, and I didn't had that issue.

richonguzman, 2024-05-16:
Yes, this is for experimental DIGI only (NOT TO [USE with WIFI + APRS-IS)