richonguzman/LoRa_APRS_iGate #97: Got TBeam v1.1 915Mhz works

N1obu, 2024-05-18:
I finally got 915Mhz to work on my TBeam v1.1
I had to modify/change the platformio.ini to use the SX1276
Can you make this change permanent in future builds?

[image]
[image]
[image]

richonguzman, 2024-05-18:
So I should add this mod as a board for using with 915 MHz ??

N1obu, 2024-05-18:
Sure. It gives others in the US options for their TBeams. There are two versions. SMA on the boards use the SX1276 and U.FL on the board use the SX1262.

wb2osz, 2024-06-12:
I would also like to see support for the SX1276.  
I tried running your application on a T-Beam v1.2 with SX1276, NEO-6M, and AXP2101.
It mostly worked.  I could configure it for my wifi LAN, and send beacons, to APRS-IS, which appeared on aprs.fi.
When I tried setting setting the LoRa frequency and enabling RX/TX, all of that stopped working.
After that, I just get the initial "LoRa APRS (IGate&Digi)" screen which does not change.
It does not attach to my wifi LAN.  It does not present its own wifi access point.
Thanks!!!

P.S.  North America, South America, Australia, and Japan allow 902-928 MHz, or a subset of that, for LoRa.

richonguzman, 2024-06-12:
Ok

Will add all versions then very soon for 915 MHz

richonguzman, 2024-06-21:
all boards with 915 version updated

closed 2024-06-21 by richonguzman