richonguzman/LoRa_APRS_iGate #98: Heltec WiFi LoRa V3.1 - Starting LoRa failed!

erikdewildt, 2024-05-19:
Hi, 

I've tried to flash a Heltec WiFi LoRa 32 V3 board (board says version 3.1) with the APRS IGate firmware.

When te board boots it immediately outputs a "Starting LoRa failed!" message and nothing more happens.

The screen shows a "LoRa APRS < IGATE & DIGI >" screen but further I can only reset the board.

Firmware on the screen shows 2024.04.13

Any help would be greatly appreciated.
Thanks,

73 de PE1PUX
Erik

richonguzman, 2024-05-19:
Hi , if you flashed this with the webInstaller (2024-04-13) , this is a known bug as the installer was created with ESP32 and not ESP32S3 as it should.

You can install it the "old way" with VSCODE to have the HELTEC V3 work fine and also have the latest improvements which are a lot (2024-05-17)

Will try to update the web installer soon , but in the meantime please do it over VSCODE

erikdewildt, 2024-05-19:
Hi, 

I've noticed the issue with the Web Flasher and started using VSCODE to flash the board. This resulted in the "LoRa Failed" error. However after a couple of more tries I got it working. When changing some settings I got the "LoRa failed" error again.

Re-did the complete flash using VSCODE and then was able to configure as desired. So I'm not sure what caused the error.

I'll close this ticket for now as it seems to work.

Thanks,
73

Erik

closed 2024-05-19 by erikdewildt