richonguzman/LoRa_APRS_iGate #99: Heltec V3 SPIFFS mount failed

MrTalon63, 2024-05-19:
Similarly to what author of the issue #98 has been experiencing. The OLED displays entry information and it hangs when trying to initilize LoRa
```
E (228) SPIFFS: mount failed, -10025
[   229][E][SPIFFS.cpp:89] begin(): Mounting��
Starting Station:    Version: 2024.05.17
Starting LoRa failed!
```
Flashed using platformio, using master branch. I have also tried erasing the flash using `pio --target erase`
my iGate config file
```json
{
    "callsign": "SP0KS-2",
    "wifi": {
        "autoAP": {
            "password": "1234567890",
            "powerOff": 10
        },
        "AP": []
    },
    "beacon": {
        "latitude": 52.493193,
        "longitude": 21.034547,
        "comment": "LoRa APRS 1k2 434.855MHz",
        "interval": 15,
        "overlay": "L",
        "symbol": "a",
        "path": "WIDE1-1",
        "sendViaAPRSIS": true,
        "sendViaRF": false
    },
    "digi": {
        "mode": 2
    },
    "tnc": {
        "enableServer": true,
        "enableSerial": false,
        "acceptOwn": false
    },
    "aprs_is": {
        "active": true,
        "passcode": "17401",
        "server": "euro.aprs2.net",
        "port": 14580,
        "filter": "m/10",
        "toRF": false
    },
    "lora": {
        "txFreq": 434855000,
        "rxFreq": 434855000,
        "spreadingFactor": 9,
        "signalBandwidth": 125000,
        "codingRate4": 5,
        "power": 30,
        "txActive": true,
        "rxActive": true
    },
    "display": {
        "alwaysOn": true,
        "timeout": 4,
        "turn180": false
    },
    "syslog": {
        "active": false,
        "server": "192.168.0.100",
        "port": 514
    },
    "bme": {
		"active": false,
		"heightCorrection": 0,
		"temperatureCorrection": 0.0
	},
    "ota": {
        "username": "",
        "password": ""
    },
    "other": {
        "rememberStationTime": 30,
        "sendBatteryVoltage": false,
        "externalVoltageMeasurement": false,
        "externalVoltagePin": 34,
        "lowPowerMode": false,
        "lowVoltageCutOff": 0
    }
}
```

richonguzman, 2024-05-19:
Hi, please install it over VSCODE as the "new" WebInstaller with the correction is not read , but the code on VSCODE is working fine.

until a few more days this is the only solution

PS: for VSCODE you need to import the project,  select the correct environment for this board, upload firmware and then upload filesystem

MrTalon63, 2024-05-19:
I see what I did wrong, I did not read the whole wiki and missed the step with uploading file system to the board.
I'm sorry for unnecessary issue, and thank you for such quick response.

closed 2024-05-19 by MrTalon63