richonguzman/LoRa_APRS_iGate #101: BME680 not working on Lilygo T3 V1.6.1

wernizh, 2024-05-23:
The LoRa_APRS_iGate does work for me on a Lilygo T3 V1.6.1
But when I do connect a BME680 to the board with 
  SDA to IO21
  SCL to IO22
and switch on  "Telemetry" then it does not boot anymore it says something like: "Telemetry is active but no sensor is found".

In this situtation it even does not connect to Wifi anymore and therefore I'm not able to switch off telemetry. So I have to start from scratch and load the firmware again...

According to the description of my board "SDA to IO21 and SCL to IO22" should be fine.
On some pages (https://github.com/richonguzman/LoRa_APRS_iGate/wiki/20.-Adding-BME280-Module) only the BM280 but on other pages the BME680 is mentioned as well.

What could I try to get it to work?

Thanks a  lot,

Werner

richonguzman, 2024-05-23:
If you installed from web installer there are a few things to consider:

1) Webinstaller (2024-04-13 version) has BME280 module as default.
2) to use bme680 you need to mod it on the code from github and install the old way with vscode 
3) luckily I have updated the code on github and it will now auto detect the module added and it should run without any modding of the code
4) I will update tomorrow the webinstaller with lots of new things including the bme680 auto discover part (which also would just warn if the discovery process fails but won't block the board to work as igate)

wernizh, 2024-05-23:
Thanks a lot. Very good/usefull project. 😃🌞

wernizh, 2024-05-23:
Thanks a lot. I did now install the firmware with vscode. And BME680 does work now. 
Super support!   🥇

closed 2024-05-23 by richonguzman