richonguzman/LoRa_APRS_iGate #102: Backup / Restore does not recover everything

wernizh, 2024-05-23:
In my configuration of the i-gate when doing backup/restore it does not recover:

- second wifi; only the first wifi will get restored  (I did not try more then two wifi)
- Longitude and latitude will be empty after restore
- some other fields for example "TX" enabled.   (I did not check all of them)

Thanks a lot, 

Werner

richonguzman, 2024-05-23:
Can you try the manual insert of the data ? Sometimes when new things are added , the restore process won't work with old restore JSON files that have less variables

wernizh, 2024-05-23:
It's not old data. Backup and restore was both today with same version of firmware.

richonguzman, 2024-05-23:
Can you test with github firmware and install it with vscode (this has the latest updates ) lot more than web installer that's gets updated tomorrow (hopefully)

S57PNX, 2024-05-23:
Could this be related to the WebUI config parsing issue that I reported in [https://github.com/richonguzman/LoRa_APRS_iGate/issues/58#issuecomment-2126659847](url) ?

richonguzman, 2024-05-23:
> I did not load the latest firmware with vscode. Still have the problem when restore the backup that for example the 2nd wifi is missing. But could also be a refresh problem of the browser. Could be that just after the restore it does not display all the values in the browser. When clicking a bit around the 2nd wifi suddenly was displayed. But I did not check further...

because of lately mods to the code the full download from github (as full and last updates are there) is needed

or use the "old" 2024-04-13 firmware from web installer

closed 2024-05-24 by richonguzman