richonguzman/LoRa_APRS_iGate #103: no position when "Send our beacon to APRS-IS" on || Lilygo T3 v1.6.1

Charly1000, 2024-05-24:
env in Visual Stuio Code I'm using: ttgo-lora32-v21

I have set the beaconing menu to "Send our beacon to APRS-IS" to on and "...to RF" to off.
It do not send the location to the APRS-IS network.

jclifford73, 2024-05-24:
See open issue #89 raised by me.  The beaconing logic is broken.    You may have the path set to blank and it will only beacon with path wide1-1 or maybe other parameters.
73 John GW4BVE

Charly1000, 2024-05-24:
None of the individual settings lead to the desired result. It doesn't work at all.

Charly1000, 2024-05-24:
RESOLVED. It works. I forgot to generate a passcode for config file: https://apps.magicbug.co.uk/passcode/index.php/passcode
so there was no connection to APRS-IS network because of that.

closed 2024-05-24 by Charly1000